fwup-home/fwup #2: Test Suit Failure

systmkor, 2015-06-22:
Compiling and running the latest commit in master branch.

## System Info
* Distro - Alpine Linux 3.2
* LibC - MUSL 
* CC - GCC 4.9.2
* Shell - ZSH 5.0.7

## Log
```
======================================
   fwup 0.3.0: tests/test-suite.log
======================================

# TOTAL: 28
# PASS:  26
# SKIP:  0
# XFAIL: 0
# FAIL:  2
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: 004_env_vars
==================

fwup: can't open file-resource '/home/admin/src/fwup/tests/work/${TEST_ENV_VAR}'
FAIL 004_env_vars.test (exit status: 1)

FAIL: 005_define
================

fwup: can't open file-resource '/home/admin/src/fwup/tests/work/${TEST_DEFINE}'
FAIL 005_define.test (exit status: 1)

```

fhunleth, 2015-06-23:
Do you know which version of libconfuse that you're using? See https://github.com/fhunleth/fwup#building. It really sounds like the libconfuse bug.

systmkor, 2015-06-23:
@fhunleth current version of confuse is 2.7 on Alpine. I take it it's a more recent revisions that fixes the bug? Should I build from their repository?

fhunleth, 2015-06-23:
@systmkor I just checked https://github.com/martinh/libconfuse, and it still doesn't look like there's a new release tag (over a year since the fix was integrated). I've submitted an issue martinh/libconfuse#23 in hopes that the maintainer will make an official release. For now, you'll need to build from their repository.

systmkor, 2015-06-23:
@fhunleth  Thanks :+1:  I tried compiling from the github libconfuse head commit however it fails (i.e. the searchpath test results in a segmentation fault) when running `make check`

fhunleth, 2015-06-23:
@systmkor Ugh. Are you running ./configure like at http://git.alpinelinux.org/cgit/aports/tree/main/confuse/APKBUILD? (Sorry - I'm just guessing right now since I haven't used Alpine Linux.)

systmkor, 2015-06-24:
@fhunleth tl;dr yes. I didn't check the configuration options since I didn't want to break other programs that depended upon that package. :clap: you google-fu :smile: I assume I need to change the ./confugre options. Also the libconfuse `make check` fails with and without those `./configure` options.

fhunleth, 2016-05-30:
Closing old issue. Hopefully fwup works on Alpine now, but if not, I would assume it to be a new issue since libconfuse is at 3.0 upstream now.

closed 2016-05-30 by fhunleth