fwup-home/fwup #3: file-resource custom path does't seem to work

CaptChrisD, 2015-06-30:
I am trying to add file to the fw bundle using the file-resource option in my fwup.conf file, like so:

```
file-resource "/release.html" {
    host-path = "${NERVES_ROOT}/../_images/release.html"
}
```

Then when the firmware bundle is made I get
```
chris@chrisdev:~/projects/ng1a$ ls _images/fw/
data  meta.conf
chris@chrisdev:~/projects/ng1a$ ls _images/fw/data/
MLO  release.html  rootfs.ext2  u-boot.img  uEnv.txt
```

From the documentation it seems like this should end up in the root of the bundle.

fhunleth, 2015-06-30:
I'm surprised that doesn't work as well. See test 2:

https://github.com/fhunleth/fwup/blob/master/tests/002_resource_subdir.test

My guess is that you're not using the latest version of fwup. I've been slow to update nerves with the latest, since I haven't used it as much, but I have no reason to believe that it wouldn't work. Sorry - really busy with non-Nerves work. :(

CaptChrisD, 2015-06-30:
nerves compiled fwup reports version 0.2.0

fhunleth, 2015-06-30:
Yep. That feature is in 0.3.0. I added that along with several other things - most notably firmware signatures. If I weren't so busy with work, I'd update nerves, but I'm just swamped at the moment. The new fwup requires libsodium for the signatures. For whatever reason, my libsodium patch is taking the slow road to be included in buildroot. Here's the patch so that you can manually add it, though: https://patchwork.ozlabs.org/patch/473879/.

Alternately, if you can handle not putting files in the root directory of the archive for a while, that might be the easiest route.

CaptChrisD, 2015-06-30:
Ok well I really just need the portion of fwup that rewrites the master boot record. So maybe I will look at how to just use that portion....

fhunleth, 2015-07-07:
Closing this issue since support for resources outside of the /data directory was added in 0.3.0.

closed 2015-07-07 by fhunleth