fwup-home/fwup #5: Minor issue with fat_setlabel

ypearson, 2015-07-05:
It seems that the label provided gets capitalized. I would be great to get a label that is exactly what is entered without alteration. I guess some upper() function gets invoked.

eg.

on-init {
        mbr_write(mbr-a)
        fat_mkfs(${BOOT_OFFSET}, ${BOOT_COUNT})
        fat_setlabel(${BOOT_OFFSET}, "boot")
    }

Actual label on SD card: BOOT

Ubuntu 14.04

fhunleth, 2015-07-07:
My guess is that this is a legacy of 8.3 DOS filenames. The upper() function can be found at https://github.com/fhunleth/fwup/blob/master/3rdparty/fatfs/src/ff.c#L3842. Since this is part of a 3rd party FAT filesystem library I use (http://elm-chan.org/fsw/ff/00index_e.html), I'm a little hesitant to change it. My quick glance through the FAT spec makes it sound like FAT volume labels are supposed to be uppercase, but I'm no expert. I'm certainly used to seeing lowercase volume labels (not sure if it's with FAT FS's or not), but I have no clue whether some ROM bootloader out there would get confused or not if it is out of spec.

fhunleth, 2016-05-30:
Closing this issue since nothing has been done on it for almost a year. It doesn't feel like support for mixed case volume labels is worth the effort to verify that it's harmless.

closed 2016-05-30 by fhunleth