fwup-home/fwup #6: Formatting FAT FS with offset past 2 GiB

edahlseng, 2015-08-05:
When running the `fat_mkfs` command to attempt to create a FAT filesystem in a FAT 32 partition, there is an error whenever the offset is exactly 2 GiB or greater.  This error seems to be independent of the size of the partition, and only depends on the offset.  The error is: `A hard error occurred in the low level disk I/O layer`.

fhunleth, 2015-08-05:
I haven't tested sizes that large, but it really sounds like I have 32-bit ints where I should have 64-bit ones. I'm pretty sure that I enabled the large file size options on the 3rd party fatfs code, but I'll have to verify and that my code is right. I'm currently very busy on a work project, but will try to get to this in the next few weeks. If you happen to figure it out before then, please send a patch and I'll try to integrate it right away.

edahlseng, 2015-08-05:
Thanks for the quick response!  I'll try to figure out a patch if I can, and will keep you updated.

fhunleth, 2015-08-24:
Closing this issue thanks to your patch.

closed 2015-08-24 by fhunleth