fwup-home/fwup #7: Unexpected error parsing meta.conf

edahlseng, 2015-08-06:
When running fwup with an upgrade task on an embedded device, an error is produced (almost immediately):
```
[buf]:66: no such option 'include-osip'
fwup: Unexpected error parsing meta.conf
```
Interestingly enough, when I run the command on my local machine with the sd card simply inserted as an external device, I don't encounter this command.

fhunleth, 2015-08-07:
The computers are running different versions of fwup. The embedded device is running an older version. If you match them up, it will work. I view this as a bug. In my own work, I tend to stick to a firmware update version and not change it (or only minorly change it) to avoid forward and backward compatibility problems, but I don't like that as a solution. The real solution is to remove libconfuse from fwup, since it adds unspecified parameters when it normalizes configurations but then rejects configurations when it sees something unknown. This can't be changed without modifying libconfuse, and this is only one of libconfuse's shortcomings (how I wish I figured this out earlier). I have an incomplete branch to fix this, but it might be a while that it can be in a form to integrate. For now, I'd suggest sticking with the same version both places.

fhunleth, 2015-12-02:
I'm going to replace this issue with #12. I can make `fwup` more robust to this backwards incompatibility problem with an upgrade to `libconfuse`. Unfortunately, they still haven't released the new version, but I hope they will soon.

closed 2015-12-02 by fhunleth