fwup-home/fwup #9: Issues running from app data partition

edahlseng, 2015-09-01:
I am attempting to run a ping-pong update with fwup.  I have a separate partition for app data and downloaded the fwup archive to that partition so that I could keep the RootFS small.  When running fwup, however, it first tries to unmount all of the partitions.  When it tries to unmount the app data partition, it receives a partition in use error, and then exits.

I think that a good solution might be to have fwup ignore any errors that it receives while trying to unmount the partitions, and to continue on.  There shouldn't be any errors due to mounted partitions, since in the configuration file we can specify if a particular partition needs to be unmounted before running.  Thoughts?

fhunleth, 2015-09-01:
Agreed. I had a particularly painful experience a few years ago with a partition not getting unmounted when it should have been, and implemented a simplistic approach that worked on my projects. A couple others have run into the same situation as you, so I'd accept a patch to ignore the error if the unmount fails. I also agree that the config file can be used to figure out the partitions that need to be unmounted.

fhunleth, 2015-09-06:
Addressed by #10.

closed 2015-09-06 by fhunleth