fwup-home/fwup #12: Upgrade to libconfuse v2.9 to improve backward compatibility

fhunleth, 2015-12-02:
The current version of libconfuse throws an error if it encounters a configuration parameter that it doesn't recognize. This is a big problem since it prevents the addition of a parameters if you want a new firmware file to be applied by an older version of `fwup`. In many cases, the new parameters are not mandatory for applying the firmware so the error is completely unnecessary. Version v2.9 of libconfuse has support for handling unknown arguments, so it will be possible to ignore these new arguments when parsing meta.conf files inside the `.fw` files. Errors should still be thrown when encountering unknown parameters when creating `.fw` files.

At this moment, libconfuse v2.9 hasn't been released. See https://github.com/martinh/libconfuse/blob/master/ChangeLog.md for details.

rmorgans, 2016-03-31:
https://github.com/martinh/libconfuse/releases/tag/v3.0

fhunleth, 2016-05-30:
libconfuse v3.0 was needed for this to work. fwup now uses the unknown parameter feature if it's available. Older libconfuse version still can be used with the old behavior, but they'll emit a warning when compiled. See 172eed535183d39f3456852645f79c176eae79e6.

closed 2016-05-30 by fhunleth