fwup-home/fwup #21: Support manipulation of ext filesystems

adamac, 2016-01-20:
In the example fw.conf (https://github.com/fhunleth/fwup/blob/master/fwupdate.conf) an ext partition is used for application data. I'd like to do exactly this but I don't see support for creating ext filesystems.

fhunleth, 2016-01-21:
fwup doesn't support ext filesystem operations. I haven't looked into how feasible this is to add. You have two options right now: 1. In your application (or in an init script), check whether the application partition is corrupt, and if it is, create it with `mke2fs`. 2. Create an ext partition image on the host and use `fwup` to write it as a blob. 

I'm not a big fan of the 2nd option, since it may take a long time to write due to zero'ing out all of the blocks in the filesystem. FWIW, I wrote `fwup` with some code to handle sparse files, but it's been so long since I've tested it, that it likely doesn't work well. In my own work, I've never had a use for including sparse files in the firmware image.

The first option is what I do. The reason is that, for whatever reason, I've had application partitions get corrupt in the field. Since I have to handle this, the logic is already in my application, so there's no reason to duplicate it in a `fwup` configuration. Additionally, I don't want to format the application partition on firmware updates. For factory programming, I've had my 'complete' image write zeros to the beginning of the application partition to ensure that it's corrupt.

mzakharo, 2016-01-21:
It would certainly be nice for fwup to have ext support, but in the mean time, I just have a multi-step process on host that looks like the following:

```bash
$DIR/fwup -a -d $DRIVE -i ${FW} -t complete
#refresh MBR
partprobe $DRIVE
echo "Format Data Partition"
PARTITION4=${DRIVE}4
mkfs.ext4  -L "data" ${PARTITION4}
sync
```

fhunleth, 2016-01-21:
I just came across this project: https://github.com/gkostka/lwext4. It looks possible to integrate into fwup, doesn't look too huge, and should be portable. If anyone wants to check into it more and give it a go, I'm not against it. In fact, if I run into situation where I need to manupulate an ext partition, I might give it a try too.

adamac, 2016-01-21:
Thanks! #2 above is what I had been doing (and it's a hassle since it can take forever to write out). #1 should be reliable but has some startup cost that I was hoping to avoid.

fhunleth, 2017-06-19:
This issue has been open for quite a long time and I don't know of anyone working on this any time soon. Going to close for now.

closed 2017-06-19 by fhunleth