fwup-home/fwup #22: Support modifying uboot environment

adamac, 2016-01-21:
My preferred way of swapping old and new partitions is to update the appropriate uboot environment variables rather than re-writing the MBR. This makes testing and debugging easier, and with a redundant uboot environment enabled should be fully power-off safe. (MBR method makes me nervous)

I've been looking at whether to add this directly to `fwup` or just incorporate it in a wrapper script. Thoughts?

mzakharo, 2016-01-21:
For u-boot support out-of-thebox: Check out this: https://sbabic.github.io/swupdate/swupdate.html  - I think both U-BOOT and MBR approaches have their usecases, and we have chosen to go with MBR approach because it is simpler, and some eMMC flash chips  support atomic-write of 512 Bytes, which is the length of MBR, making it play rather nicely together.

adamac, 2016-01-21:
I've seen that project; this feature aside, `fwup` feels like the better option. I'm specifically concerned with environments that don't use eMMC and as such the MBR operation is not power-off safe.

I don't think it would be much work to add an action to set uboot environment variables. I'm considering making the additions. This could be an alternative to the MBR approach.

mzakharo, 2016-01-22:
This contribution would also benefit projects that do not use MBR,  like some Intel embedded platforms!

fhunleth, 2016-01-22:
I completely agree that this would be a useful addition.

fhunleth, 2016-10-06:
I finally added U-boot environment support. This isn't a complete implementation of all of the U-boot environment options (redundant env, big endian support are missing). It is good enough, though, for me to replace the use of `mkenvimage` and `raw_write` in my projects that were modifying the U-Boot environment. Regression test 082 has an example of A/B firmware updates using U-boot environment settings rather than modifying the MBR. The commit is https://github.com/fhunleth/fwup/commit/8e40088d3377fe8dceccc94691f606088c8e4bcc.

closed 2016-10-06 by fhunleth