fwup-home/fwup #24: Unable to install libsodium-dev in the proposed way

bordeltabernacle, 2016-02-29:
On Ubuntu 14.04 installing libsodium-dev did not work for me, I got an `unable to find package` result.

I installed it from [this](https://gist.github.com/jonathanpmartins/2510f38abee1e65c6d92) instead.

```bash
sudo add-apt-repository ppa:chris-lea/libsodium
sudo apt-get update && sudo apt-get install libsodium-dev
```

fhunleth, 2016-02-29:
Thanks for figuring this out and letting me know. Do you know how official that ppa is?

bordeltabernacle, 2016-03-01:
I don't, but in a quick google search it turns up in a few places, looks pretty legit, but I don't know a lot about these things.

https://launchpad.net/~chris-lea/+archive/ubuntu/libsodium
https://github.com/travis-ci/apt-source-whitelist/pull/145
https://paragonie.com/book/pecl-libsodium/read/00-intro.md

bordeltabernacle, 2016-03-01:
It also appears as though `libarchive-dev` is now available through apt-get.

fhunleth, 2016-05-13:
Closing this since `.deb` files are provided now. The `.deb` has a statically linked `fwup` that doesn't require libsodium or libarchive to be installed.

closed 2016-05-13 by fhunleth