fwup-home/fwup #25: Error when running ./configure in Vagrant

bordeltabernacle, 2016-03-01:
In a Vagrant Ubuntu Trusty64 box, when running the `./configure` command inside the fwup directory I get the following error:

    ./configure: line 12447: syntax error near unexpected token `BASH_COMPLETION,'
    ./configure: line 12447: `    PKG_CHECK_MODULES(BASH_COMPLETION, bash-completion >= 2.0,'

I have tried uninstalling and reinstallling bash-completion but this had no effect.  I didn't have this problem  on my host laptop, which is Xubuntu 14.04.

Sorry I can't be more help, I really don't know Bash.

fhunleth, 2016-03-01:
That's with master. I'll look into fixing that, but I'd recommend trying the latest release. I only just added bash completion support.

fhunleth, 2016-03-01:
Also, I fixed the error with determining the bash completion installation directory, so master should work as well now. I'd still stick with v0.6.0 (or subsequent releases) for your Vagrant provisioning script, though, so that you always hitting a known good version. Thanks for reporting the issue.

Radivarig, 2019-08-15:
Hi, comming from another package but with same error message. Based on [this answer](https://stackoverflow.com/a/36659975/3754246) I had to install `pkgconfig`.

fhunleth, 2019-08-16:
@Radivarig It's been a while since I've used Vagrant. Is it still a thing where projects include a Vagrant configuration or snippet to help people use it? If so, would you mind sharing something that I could put into the documentation for others?

Radivarig, 2019-08-16:
Sorry for the necropost, I've not used Vagrant so far but this issue was
one of the two results I found for this particular error so I just shared
the solution. It was not obvious to me that the problem had nothing to do
with bash-completion but instead, as the error says, syntax part before it,
which becomes valid after the installation of `pkgconfig`.