fwup-home/fwup #29: eject fails on OSX with unhelpful error message

fhunleth, 2016-05-29:
Here's the error:

 `fwup: eject failed: 0xc010 (49168) (null))`

This was reported by @wsmoak on the elixir-lang slack. I'm not sure if improvements to the error message would help debug this particular issue. Check if error 49168 could be a commonly encountered error if the SDCard is busy and produce a better error message.

rkenzhebekov, 2016-06-23:
@fhunleth, I got this same error:
```
fwup: eject failed: 0xc010 (49168) (null))
```

fhunleth, 2016-06-24:
I'm not a Mac person, but if anyone has any clues to this error or what's happening, please share. I also can't reproduce it.

fhunleth, 2016-09-08:
After some googling, it sounds like people are getting around this error by disabling their anti-virus software. I'm wondering if those of you who've seen this have an anti-virus program running.

wsmoak, 2016-09-09:
I turned off Bitdefender, but I still get this (with fwup 0.8.0)

```
$ mix firmware.burn
Use 3.71 GiB memory card found at /dev/rdisk2? [Yn] y
100%
fwup: eject failed: 0xc010 (49168) (null))
```

fhunleth, 2016-09-09:
Bummer. Thanks for letting me know. I guess it's time for more googling. If anyone knows what error 49168 is, that would be helpful. I can't find it.

fhunleth, 2017-06-23:
Since it seems like this error is harmless, I silenced it in 8dbc0ca01f038123263b8ea8601378d0493b6a07. I never could find information about the error.

fhunleth, 2017-06-29:
Updates: 

It turns out that 0.15.0 caused the error to hit more often. I found and fixed an issue and released 0.15.1. I re-enabled the error message in 0.15.1 since I thought that this issue was really fixed now. 

Last night I witnessed an eject failure with 0.15.1. The person who had the laptop said that it was in really poor shape and had many issues with USB devices. She was using a USB Multicard reader to program the MicroSD card, so that may be related. Her SDCard slot didn't work. She insisted that the eject problem was likely due to her laptop. However, the error definitely was happening, so reopening...

fhunleth, 2020-04-30:
I haven't seen this in a long time. I'm not sure what changed. I didn't do anything though.

closed 2020-04-30 by fhunleth