fwup-home/fwup #42: Not compiling because of help2man

philipgiuliani, 2016-09-13:
I had issues compiling the 0.9.0 version, while 0.8.2 works fine.

`./configure` runs fine, but `make` fails with the following error:
```
make[2]: *** No rule to make target 'fwup.h2m', needed by 'fwup.1'.  Stop.
```

After removing the following code from `src/Makefile.am` it worked fine.
```
if HAVE_HELP2MAN
man_MANS = fwup.1
CLEANFILES = $(man_MANS)
EXTRA_DIST += fwup.h2m

fwup.1: fwup$(EXEEXT) fwup.h2m
	$(HELP2MAN) --no-discard-stderr --output=$@ --no-info --include fwup.h2m \
	    -n "configurable embedded firmware update creator and runner" \
	    ./fwup$(EXEEXT)
else
	@echo  "Warning: help2man not available, no man page created."
endif
```

If you have any questions, just ask!

fhunleth, 2016-09-13:
I see that there's an issue since `EXTRA_DIST += fwup.h2m` should be outside of the `if` statement. However, I'm not sure that this is causing your issue. Do you have this file? https://github.com/fhunleth/fwup/blob/master/src/fwup.h2m. If you don't, then how did you get the fwup source code? Git, releases link, somewhere else?

philipgiuliani, 2016-09-13:
Seems that i dont have the file. Downloaded it from the Github releases page!

fhunleth, 2016-09-13:
I found it. Travis didn't have help2man installed and the `if` statement above caused the file to not be included. I could have sworn that travis had help2man installed for earlier releases.

I've pushed two changes to fix this issue: c077698b2386e81838972810c950a9a2dc6a9670 and e4abcc2c6f5bf8252ad0486d87528149d7fd8094. The first will always distribute the fwup.h2m. The second makes sure that help2man is run on Travis so that manpage generation is exercised.

Thanks for letting me know about this!

closed 2016-09-13 by fhunleth