fwup-home/fwup #44: Test 030 and 032 failing on raspbian

philipgiuliani, 2016-09-13:
The following two tests are failing on raspian. I looked at them but could not find out why. On my SD Card i have 20 GB of space left.
* tests/030_2T_offset.test
* tests/032_fat_2T_offset.test

fhunleth, 2016-09-13:
Those tests fail on systems that don't support sparse files. They actually don't use up much disk space at all. I assume that Raspbian uses ext4 so it *should* be fine, but maybe there's another constraint that kicks in on 32-bit processors.

philipgiuliani, 2016-09-13:
Here the output of `df -T`, as you said, its using ext4.

```
Filesystem     Type     1K-blocks    Used Available Use% Mounted on
/dev/root      ext4      30584708 2622588  26689956   9% /
devtmpfs       devtmpfs    493904       0    493904   0% /dev
tmpfs          tmpfs       498236       4    498232   1% /dev/shm
tmpfs          tmpfs       498236    6656    491580   2% /run
tmpfs          tmpfs         5120       4      5116   1% /run/lock
tmpfs          tmpfs       498236       0    498236   0% /sys/fs/cgroup
/dev/mmcblk0p1 vfat         64456   20984     43472  33% /boot
```

Just tell me when i can do something to help you identifying the issue.

fhunleth, 2016-09-20:
I commented these out since I'm getting errors from the OS. It doesn't look like a `fwup` issue. See https://github.com/fhunleth/fwup/commit/6df3536ad0f6720d23ff577460c66f0fa100c040.

closed 2016-09-20 by fhunleth