fwup-home/fwup #45: fwup_0.9.0_amd64.deb package is rejected by Ubuntu 14.04

jmedding, 2016-09-14:
When trying to install **fwup_0.9.0_amd64.deb** I get an error: **This package is of bad quality**

> Lintian check results for /tmp/mozilla_jon0/fwup_0.9.0_amd64.deb:
> E: fwup: maintainer-name-missing <travis@testing-gce-c23388e7-daf4-4bba-91d1-7b6fcd0520f9>
> E: fwup: maintainer-address-malformed <travis@testing-gce-c23388e7-daf4-4bba-91d1-7b6fcd0520f9>

I checked version 0.8.2 and it's the same message, although the values in the details are different.
 couldn't find anything on the googles, so I just installed it.  It seems to work fine.  If this is normal, just close this ticket.  Thiss issue might help others with the same thing though.

fhunleth, 2016-09-14:
Thanks for the report! I fixed both of those errors and a couple more in a2b791b9d7f47cd16967d06defe70349c964318e. Lintian still complains about more things for me (Ubuntu 16.04), but those will take more time. None of the errors hurt installation, though.

fhunleth-lkc, 2016-09-15:
I took another pass at `fpm` to generate a better `.deb` file (https://github.com/fhunleth/fwup/commit/29599930f75ab35523322ac7b13ff6608ab7d2ce). I think that I'm about as close as can be achieved with `fpm`. This doesn't pass the version of Lintian on my system, but I don't receive any errors or warnings when I install it.

closed 2016-09-15 by fhunleth-lkc

jmedding, 2016-09-16:
I can confirm that everything works (rpi3 is blinking happily) even with the error messages.  Thanks for responding and improving.

fhunleth, 2016-09-16:
Great! Thanks for reporting back.