fwup-home/fwup #48: Add possibility to exclude certain devices

philipgiuliani, 2016-09-19:
**Problem**
On a Raspberry the system sdcard is suggested instead of a external attached sd card.

**Possible solutions**

1. `~/.fwup.conf` file with a list of interfaces to exclude/include

2. `~/.fwup.conf` which allows to set all the defaults of the argument i can pass (so i dont have to specify `-d` everytime)

3. Maybe the auto-detect feature should show all attached sd cards and you can set a default (which than would write the config):
```
Which memory card would you like to use (press d to set as default)?
[*] 29.72 GiB at /dev/mmcblk0
[ ] 7.14 GiB at /dev/sda
```

fhunleth, 2016-09-20:
@mobileoverlord - FYI: this issue may affect the Nerves tooling. I'm still thinking of a solution. The backstory to this issue is that @philipgiuliani is running on a Raspberry Pi where the SDCard containing the OS looks a lot like an SDCard that `fwup` would like to program. Use of a configuration file came up while we were brainstorming possible solutions. I'm currently leaning towards being smarter with the auto-detect logic for this situation. That doesn't mean there wouldn't be a conf file, though. Also, I believe that we're swapped in how we ask the user to proceed. `fwup` defaults to not burning an auto-detected SDCard, and `mix firmware.burn` defaults to burning. We should probably make it consistent.

GregMefford, 2016-09-20:
You could process the output of `df` to ignore the drive that's mounted at `/`.
I was thinking you could ignore anything that's in `df`, since the person wouldn't be mounting their SD card, but that's probably not as robust with Linux auto-mounting USB drives.

fhunleth, 2016-09-21:
@philipgiuliani: I added detection of which drive is mounted at `/` to exclude it from the autodetect list. It's in the `autodetect` branch. Could you try it out? You can get a prebuilt `.deb` package for Raspbian here: http://files.troodon-software.com/fwup/autodetect/fwup_0.9.2-dev_armhf.deb. It will install `fwup` to `/usr/bin`, so if you've previously compiled from source, be sure to remove the one in `/usr/local/bin`.

philipgiuliani, 2016-09-21:
@fhunleth I'm not at home, i'll try it out today evening.

philipgiuliani, 2016-09-21:
It seems to have worked fine.

```bash
pi@philipdev:~/workspace/gardex/apps/fw $ mix firmware.burn
Use 3.69 GiB memory card found at /dev/sda? [Yn]
```

Awesome, thanks! :)

fhunleth, 2016-09-21:
Great! Thanks for verifying. I'll merge the change to `master` for the next release.

I'm going to keep this issue open a little bit longer since the config file is still interesting. It has some ramifications, though, that need thought. If I don't get any great ideas before the next release, I'll close this and possibly open an issue solely regarding the config file.

fhunleth, 2016-10-23:
I'm closing this issue, since I'm pretty sure that the original problem has been addressed. I'm not opposed to having a global config file, but I don't feel like there's enough motivation to spend time adding it (for me at least).

closed 2016-10-23 by fhunleth