fwup-home/fwup #50: [WIN 10]fwup burning to SD card not working as expected

gallexme, 2016-09-20:
Log:
fwup -a -i rpi/blinky.fw -t complete
Use 1.89 GiB memory card found at \\.\PhysicalDrive2? [y/N] y
100%
they were never bootable by rpi
and sometimes the .txt files were "jibberish"(broken)
example cmdline.txt: ÂœV¾ƒã|“nÙÄ¶õ[á157T•§ñ´6öœ§ lÑúÆeá*iûÚ–))Žœ(-¶æPÜ»oÕç²º¾È

but writing to .img file then writing it with win32 disk imager to the sd card it works

fhunleth, 2016-09-20:
@GregMefford or @michaelkschmidt, have you seen this?

michaelkschmidt, 2016-09-20:
I have not.  @gallexme can you send us the .fw file to see if I can reproduce it?

Also, when you write to the .img file, are you doing that in Docker or Windows?

gallexme, 2016-09-20:
>Also, when you write to the .img file, are you doing that in Docker or Windows?
windows, im just creating the .fw file on a linux docker everything else on windows
[blinky.zip](https://github.com/fhunleth/fwup/files/483362/blinky.zip)

i dont think that there is anything special about the .fw

GregMefford, 2016-09-21:
I didn't have a chance to look into this yet, but I'm curious if it's related to the block size of the SD card. Do you by chance have a different model of card you can experiment with?

fhunleth, 2016-09-21:
@GregMefford I'm not aware of any non-512 byte block size SDCards. There are 4K hard drives, but not SDCards. If you're referring to erase block sizes, the SDCard firmware handles those variations.

This issue feels more to me like the drive didn't unmount all the way when fwup was writing to it. Then, Windows wrote some things on top of what fwup wrote and the contents were out of sync.

gallexme, 2016-09-21:
@fhunleth gonna test that later (fully unmounting it )

michaelkschmidt, 2016-09-21:
@fhunleth the block size was an issue during the transition from SD -> SDHC.  There were a few cards manufactured that 1K Blocks to reach 2 GB.  

Some more info here: https://en.wikipedia.org/wiki/Secure_Digital#Transfer_modes

@gallexme what size card are you using?

gallexme, 2016-09-21:
@michaelkschmidt  1gb or 2gb not sure(not home) but im 90 % sure 2gb

michaelkschmidt, 2016-09-21:
OK, I think I have reproduced the issue

Looks like there are a couple bugs:
- [ ] We are not unmounting the FAT partition, so it does not get updated 
- [ ] UBoot Stage1 doesn't appear to be writing

That said, if you have already written the card once, both items are already there and fwup can update the rest.

@gallexme can you confirm that if you write the card once with win32 disk imager, that you can update the app just using fwup on subsequent attempts?

fhunleth, 2016-09-21:
If the FAT partition isn't unmounted, then all bets are off. I don't think there's any point in experimenting until that's fixed. Since the FAT partition is always updated on Raspberry Pis, even the smaller upgrade tasks aren't guaranteed to work. I think that if they do work that it's just luck since the files being upgraded are most likely not changing.

gallexme, 2016-09-21:
@michaelkschmidt  actually i cant update 
fwup -a -i rpi/fw.fw -t upgrade
Use 1.89 GiB memory card found at \\.\PhysicalDrive2? [y/N] y
 21%
fwup: fat_write can't write(zImage.new): A hard error occurred in the low level disk I/O layer

michaelkschmidt, 2016-09-21:
@gallexme try the "complete" task:
```
fwup -a -i rpi/blinky.fw -t complete
```
The only reason I suggest this is that it has been accidentally working for me :disappointed:

closed 2016-09-22 by fhunleth

fhunleth, 2016-09-22:
@gallexme: GitHub auto-closed this issue, but it really looks like @michaelkschmidt fixed it. Could you try out one of these and see if his fix works for you?

http://files.troodon-software.com/fwup/master/fwup.exe
http://files.troodon-software.com/fwup/master/fwup.0.9.2-dev.nupkg

gallexme, 2016-09-23:
@fhunleth seems to work 👍

fhunleth, 2016-09-23:
@gallexme: Thanks! I'm generating a release with the fix now. Thank you very much for the report!

PR #51 (2016-09-22): Windows: Keep Logical Volumes locked while writing
This fixes #50