fwup-home/fwup #54: Unable to raw_write single resource to two different offsets

robhemsley, 2017-01-13:
When specifying the same resource to be written to two different offsets it fails with the following message.

> fwup: raw_write didn't write anything. Was it called twice in an on-resource for 'rootfs.ext4'?

In my setup both partitions should have the same resource written during factory setup. I would like to avoid having to increase the size of my .fw bundle with a duplicate resource. My config currently looks like this:

```
    on-resource rootfs.ext4 {
        raw_write(${ROOTFS_A_PART_OFFSET}) # Write to the first rootfs partition
        raw_write(${ROOTFS_B_PART_OFFSET}) # Write to the second rootfs partition
    } 
```

Is this an intentional design decision?

fhunleth, 2017-01-13:
The intentional design decision was to make `fwup` streamable for updates (i.e. no disk space is required to hold the .fw file on device). Everything flows down from that decision. That's why you can't write a resource twice, since it would either require caching the whole resource or rewinding. In theory, there's nothing preventing a function like `raw_write2` that writes the file to two places while it's being streamed through, but it starts getting weird.

I have a couple ideas for you:
1. Call `fwup` twice to create the factory image. You'll need two tasks to do this.
2. Don't write rootfs B in the factory.

In all of my setups, I've convinced myself that if the B partition ever gets loaded from a clean factory install on A, then something is really wrong that needs to be investigated. I actually intentionally write zeros to the beginning of the B partition in the factory image so that I know it is invalid just in case someone is refurbishing a device that didn't make it through manufacturing the first time.

Also, I'm not set against figuring out a way to support a multiple write feature, so let me know your use case and maybe we can figure something out.

robhemsley, 2017-01-20:
That makes complete sense. This would only be need in the factory and so I'm happy to just create two tasks and run fwup multiple times during setup. 

Thanks for taking the time to explain.

closed 2017-01-20 by robhemsley