fwup-home/fwup #57: fwup.1 man page contains error message

Polynomial-C, 2017-02-21:
Hi,

the fwup.1 man page generated by help2man contains the following line in the DESCRIPTION section:

```
./fwup:  unrecognized  option '--help'
```
The reason is help2man calling `fwup --help` which fwup does not understand. Unfortunately help2man refuses to call a binary with no `--help` option at all (adding `--help-option=""` in src/Makefile.am makes help2man fail).

fhunleth, 2017-02-21:
Thanks! I can't believe that I hadn't noticed that when I first added man page generation, but it's definitely there. Fix coming shortly...

Polynomial-C, 2017-02-21:
Okay, seems like you beat me in fixing this ;)

fhunleth, 2017-02-21:
Oops - I should have waited. The bug was a little embarrassing, though, so I wanted to be done with it. I appreciate you taking the time to let me know and for putting together the patch.

Polynomial-C, 2017-02-21:
Nah, it's all fine. From a quick glance your patch seems to be more elegant than mine. My C is quite rusty anyway.
Am I right that your patch returns failure when invoking --help? Shouldn't that rather be success because when `fwup --help` prints the help message it successfully did what the user requested?
Oh and out of curiosity, why not using the `-h` short option but `-?`?

fhunleth, 2017-02-21:
You're right - mine does return error when specifying `--help` and that's not right. As for `-h` vs `-?`, my fingers type `-?` frequently, but `-h` makes sense too. Would you mind resubmitting your patch with the one simplification of handling the call to print_usage in the switch rather than making a full-fledged command out of it? You really should get full honors for this.

Polynomial-C, 2017-02-21:
Here we go. I kept `-h` as short option but would not mind to change that to `-?` if you prefer that.

fhunleth, 2017-02-21:
Looks good to me. I merged it. Thanks again!

closed 2017-02-21 by fhunleth

Polynomial-C, 2017-02-21:
D'uh... one minor styling issue with my patch... line 331 is not correctly aligned.

fhunleth, 2017-02-21:
No worries - I'll fix it.