fwup-home/fwup #60: Sparse file tests fail on ZFS

georgewhewell, 2017-05-19:
I can replicate this on ZFS, but all pass on ext4.

https://gist.github.com/Mic92/cda72ea4f028ff33e440c22dc2df15de

fhunleth, 2017-05-19:
Thanks for the report on ZFS! I tried it out. ZFS has a minimum sparse hole size of 128 KB vs. ext4's 4 KB. The failed unit tests rely on the minimum sparse hole size. Since it's completely reasonable for a filesystem to limit the minimum hole size, I modified the regression tests to verify it at runtime. This makes all tests pass on ZFS for me (and cleans up some code). 

My changes are at: https://github.com/fhunleth/fwup/tree/test-sparse

If it's easy for you to try, could you let me know if that branch works for you? 

Thanks!

georgewhewell, 2017-05-20:
@fhunleth Very nice, thanks- all sparse tests now pass. Test 105 still fails on my ZFS machine (and NixOS builder- https://hydra.nixos.org/build/53010322), although it passes on my ext4 machine.

fhunleth, 2017-05-20:
I may need to set up a ZFS machine to debug, but I think that I have a guess. Test 105 exercises code that detects where the root filesystem is mounted. To do this, it tries to figure out the right answer in a shell script, but I don't see how this works if the root filesystem is on more than one disk. Since ZFS has volume management built in, I wonder if it won't work even with one disk. 

Anyway, I put a check it to detect that situation, print a message and skip the test. That's in this same branch. If this doesn't fix test 105 for you, could you look at the `105_require_path_on_device.log` file and run `stat / -c "%04D"`?

As a side comment, I had never tried out ZFS or looked around NixOS much at all. Both are really cool!

georgewhewell, 2017-05-20:
All tests pass on this branch now, many thanks.

On one ZFS machine I get `0015`, on another I get `0016`.

fhunleth, 2017-05-20:
Great! I merged the changes to `master`. I'll be making a patch release later this afternoon so that your NixOS package can use a proper release version.

Thanks again!

closed 2017-05-20 by fhunleth