fwup-home/fwup #67: Support writing to SDCards from within the Windows Subsystem for Linux

fhunleth, 2017-08-01:
With some work, it's possible to get `fwup` to both run in the Windows Subsystem for Linux and be able to write to SDCards. Here's a report from @jmerryweather:

steps are as follows:
(1) uninstall any linux version of fwup in WSL
(2) from Windows install fwup using chocolatey
(3) create a 'fwup' file that will will point to fwup.exe, i can send you the version i made, it is based on the VS Code 'code' file
(4) run Bash on Ubuntu on Windows "As Administrator", fwup won't work without doing that, it will instead give errors that there are no sd cards
(5) without 'sudo', run the following: fwup firmware.fw

 ```
#!/usr/bin/env bash
#
# Inspiration taken from Microsoft VS Code's Windows 'code' bash file

NAME="fwup"
FWUP_LOCATION="$(dirname "$(realpath "$0")")"

FULLPATH="$FWUP_LOCATION/$NAME.exe"
if grep -q Microsoft /proc/version; then
    # If running under WSL
  "$FULLPATH" "$@"
    exit $?
fi
exit $?
```

thats the fwup file i put under C:\ProgramData\chocolatey\bin

if we could get the above distributed with the fwup chocolatey package, it would make discovery of the windows fwup much easier

fhunleth, 2017-08-01:
@michaelkschmidt - What do you think about adding the above script to Chocolatey package? Do you see any issues?

michaelkschmidt, 2017-08-01:
@fhunleth This looks like a good start.  We don't want to break fwup running under `msys` bash, though.  How does this look?
```
#!/usr/bin/env bash
#
# Inspiration taken from Microsoft VS Code's Windows 'code' bash file

NAME="fwup"
FWUP_LOCATION="$(dirname "$(realpath "$0")")"

FULLPATH="$FWUP_LOCATION/$NAME.exe"
if grep -q Microsoft /proc/version; then
   # If running under WSL
 "$FULLPATH" "$@"
else
  fwup.exe "$@"
fi
exit $?
```

fhunleth, 2017-08-01:
Good point on `msys`.

Also, I haven't run this, but shouldn't `"$FULLPATH" "$@"` work for all cases? I'm trying to think about when `fwup.exe` would be someplace other than `$FWUP_LOCATION/fwup.exe`. 

Could the script just be:

```
#!/usr/bin/env bash

FWUP_LOCATION="$(dirname "$(realpath "$0")")"
"$FWUP_LOCATION/fwup.exe" "$@"
```

Also, I dropped the `exit $?` since shell scripts return the exit status of the last program run by default. However, if Windows needs this, then please add back.

jmerriweather, 2017-08-01:
I've tested the above more concise fwup wrapper under WSL, cmd and powershell. It works well. It doesn't look like Windows needs the `exit $?`

fhunleth, 2017-08-02:
Great! @jmerriweather Do you want the honors of submitting the PR? Sounds like the script just needs to be put in the scripts directory and a line added to copy it over here: https://github.com/fhunleth/fwup/blob/master/scripts/build_pkg.sh#L59.

jmerriweather, 2017-08-02:
Sure! I've submitted the PR

michaelkschmidt, 2017-08-03:
I just tested and the simplified script seems to still work with `msys`.

michaelkschmidt, 2018-08-08:
I just realized this issue is still open

@jsiems added a workaround in `mix firmware.burn`.  See https://github.com/nerves-project/nerves/pull/288

Do we still want a `fwup` only solution?

fhunleth, 2018-10-08:
I think we can close this. If someone needs this outside of Nerves, we can point them to what was done there. My hunch is that `fwup` has very few WSL users outside of those using Nerves.

closed 2018-10-08 by fhunleth