fwup-home/fwup #80: make check on OpenBSD fails (15/140)

niko, 2017-11-12:
```
=======================================
   fwup 0.17.0: tests/test-suite.log
=======================================

# TOTAL: 140
# PASS:  125
# SKIP:  0
# XFAIL: 0
# FAIL:  15
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: 038_write_15M
===================

|                                    | 1% (0.20 / 15.00) MBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-038_write_15M.test'
FAIL 038_write_15M.test (exit status: 134)

FAIL: 039_upgrade
=================

|=================================== | 99% (152.56 / 152.56) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-039_upgrade.test'
FAIL 039_upgrade.test (exit status: 134)

FAIL: 045_legacy_require_partition1
===================================

|=================================== | 99% (152.56 / 152.56) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-045_legacy_require_partition1.test'
FAIL 045_legacy_require_partition1.test (exit status: 134)

FAIL: 052_file_concatenation
============================

|==============================      | 85% (130.70 / 152.05) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-052_file_concatenation.test'
FAIL 052_file_concatenation.test (exit status: 134)

FAIL: 073_multistep_fat
=======================

|                                    | 1% (0.15 / 15.16) MBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-073_multistep_fat.test'
FAIL 073_multistep_fat.test (exit status: 134)

FAIL: 074_fat_cache_fail
========================

|=========                           | 25% (1.18 / 4.58) MBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-074_fat_cache_fail.test'
FAIL 074_fat_cache_fail.test (exit status: 134)

FAIL: 075_big_fat_fs
====================

|=                                   | 4% (0.66 / 15.00) MBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-075_big_fat_fs.test'
FAIL 075_big_fat_fs.test (exit status: 134)

FAIL: 082_uboot_upgrade
=======================

|=================================== | 99% (152.56 / 152.56) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-082_uboot_upgrade.test'
FAIL 082_uboot_upgrade.test (exit status: 134)

FAIL: 085_raw_memset
====================

|=================================== | 99% (259.58 / 262.14) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-085_raw_memset.test'
FAIL 085_raw_memset.test (exit status: 134)

FAIL: 095_sparse_apply
======================

fwup: Unsigned archive '/home/niko/tmp/fwup/tests/work-095_sparse_apply.test/fwup.fw' is not corrupt.
/home/niko/tmp/fwup/tests
fwup: Unsigned archive '/home/niko/tmp/fwup/tests/work-095_sparse_apply.test/new.fw' is not corrupt.
|=================================== | 99% (7.17 / 7.17) KBpthread_mutex_destroy on mutex with waiters!
pthread_mutex_destroy on mutex with waiters!
|====================================| 100% (7.17 / 7.17) KB
Success!
Elapsed time: 0.003s
|===                                 | 9% (0.84 / 9.24) MBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-095_sparse_apply.test'
FAIL 095_sparse_apply.test (exit status: 134)

FAIL: 109_includes
==================

|===============================     | 87% (130.70 / 150.00) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-109_includes.test'
FAIL 109_includes.test (exit status: 134)

FAIL: 110_compression_level
===========================

|=========                           | 26% (128.78 / 493.57) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-110_compression_level.test'
FAIL 110_compression_level.test (exit status: 134)

FAIL: 111_streaming_exit_fast
=============================

./111_streaming_exit_fast.test[27]: timeout: not found

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-111_streaming_exit_fast.test'
FAIL 111_streaming_exit_fast.test (exit status: 127)

FAIL: 125_trimmed_upgrade
=========================

|=================================== | 99% (152.57 / 152.57) KBAbort trap (core dumped) 

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-125_trimmed_upgrade.test'
FAIL 125_trimmed_upgrade.test (exit status: 134)

FAIL: 131_raw_key
=================

Firmware signing keys created and saved to fwup-key.pub and fwup-key.priv

Distribute fwup-key.pub with your system so that firmware updates can be
authenticated. Keep fwup-key.priv in a safe location.
43+0 records in
43+0 records out
43 bytes transferred in 0.000 secs (394495 bytes/sec)
86+0 records in
86+0 records out
86 bytes transferred in 0.000 secs (601399 bytes/sec)
/home/niko/tmp/fwup/tests
pthread_mutex_destroy on mutex with waiters!
pthread_mutex_destroy on mutex with waiters!
fwup: Signed archive '/home/niko/tmp/fwup/tests/work-131_raw_key.test/fwup.fw' is not corrupt. Pass a public key to verify the signature.
pthread_mutex_destroy on mutex with waiters!
pthread_mutex_destroy on mutex with waiters!
pthread_mutex_destroy on mutex with waiters!
pthread_mutex_destroy on mutex with waiters!
pthread_mutex_destroy on mutex with waiters!
pthread_mutex_destroy on mutex with waiters!
fwup: Error reading or decoding public key from file '/home/niko/tmp/fwup/tests/work-131_raw_key.test/fwup-key.pub.unpadded'

Test failed!

Leaving test work files in '/home/niko/tmp/fwup/tests/work-131_raw_key.test'
FAIL 131_raw_key.test (exit status: 1)
```

fhunleth, 2017-11-12:
What version of OpenBSD are you running? I don't regularly run OpenBSD, so if there's anything else I should know to reproduce these issues, could you let me know that as well.

Also, if you happen to know whether there's an OpenBSD CI server that I can use to automate testing of `fwup` I would most appreciate it. 

As a final note, if you're just creating .fw files on OpenBSD, you should be fine even with those test failures. The failures all look like they're verifying the more rarely used features.

fhunleth, 2017-11-15:
Ok, I went through and fixed all of the OpenBSD failures. This was unbelievably helpful as it pointed out several subtle, but legit issues. Thanks for bringing this to my attention!

commit 60ce1ee3 (2017-11-15): Switch thread sync to condition variables

This fixes an issue where the cache depended on undefined behavior from
pthread_mutex to work correctly. It was called out by OpenBSD. Thanks
OpenBSD.

Partially fixes #80.
commit bec01243 (2017-11-15): Skip test 111 on the BSDs

This test requires the timeout command from the GNU coreutils package.
The alternatives look involved to implement, so skip this for now.
Luckily, this test really only makes sense on target devices and is
tested regularly on OSX and Linux.

Partially fixes #80.
commit c229e34e (2017-11-15): Fix missing NULL terminator

Thank you OpenBSD for catching this!

Partially fixes #80.
closed 2017-11-15 by fhunleth