fwup-home/fwup #83: make check failing on NixOS

jtojnar, 2017-12-01:
We first encountered the test failures when building

<details><summary>Test suite log for 0.15.0</summary>

```
=======================================
   fwup 0.15.0: tests/test-suite.log
=======================================

# TOTAL: 125
# PASS:  56
# SKIP:  0
# XFAIL: 0
# FAIL:  69
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: 001_simple_fw
===================

  0%
 99%
100%
Elapsed time: 0.000s
verify-syscalls: open was never called on 'fwup.img'

Test failed!

Leaving test work files in '/tmp/nix-build-fwup-0.15.0.drv-0/source/tests/work-001_simple_fw.test'
FAIL 001_simple_fw.test (exit status: 1)

FAIL: 002_resource_subdir
=========================

  0%
 33%
 66%
 99%
100%
Elapsed time: 0.000s
verify-syscalls: open was never called on 'fwup.img'

Test failed!

Leaving test work files in '/tmp/nix-build-fwup-0.15.0.drv-0/source/tests/work-002_resource_subdir.test'
FAIL 002_resource_subdir.test (exit status: 1)

FAIL: 003_write_offset
======================

  0%
 43%
 87%
 99%
100%
Elapsed time: 0.001s
verify-syscalls: open was never called on 'fwup.img'

Test failed!

Leaving test work files in '/tmp/nix-build-fwup-0.15.0.drv-0/source/tests/work-003_write_offset.test'
FAIL 003_write_offset.test (exit status: 1)

FAIL: 004_env_vars
==================

  0%
 99%
[... 1270 lines cut ...]
-- verify-syscalls report --
open() was never called on 'fwup.img'!
----------------------------

Test failed!

Leaving test work files in '/tmp/nix-build-fwup-0.18.0.drv-0/source/tests/work-052_file_concatenation.test'
FAIL 052_file_concatenation.test (exit status: 1)

FAIL: 053_framed_progress
=========================

-- verify-syscalls report --
open() was never called on 'fwup.img'!
----------------------------

Test failed!

Leaving test work files in '/tmp/nix-build-fwup-0.18.0.drv-0/source/tests/work-053_framed_progress.test'
FAIL 053_framed_progress.test (exit status: 1)

FAIL: 060_framed_streaming
==========================

```

fhunleth, 2017-12-01:
Thanks for letting me know about this. That really looks like an issue with the test suite rather than fwup itself. Between the versions that you referenced, I added code to validate syscalls since there are non-obvious constraints with buffer alignment, sizes and offsets on some platforms. That code is failing, but it's failing since the syscalls aren't being captured. I doubt that anything here is a problem for NixOS, but something could be hiding.

I'm not a NixOS user so I tried pulling down the NixOS appliance and installing everything needed to build fwup (like gcc, autoconf, automake, libconfuse, libarchive, etc.). I'm stuck at what looks like a noob issue, but my searches are failing me. `gcc` fails to build anything with `error trying to exec 'as': execp: No such file or directory`. I didn't see a `build-essential` package for NixOS. Any ideas how I may reproduce and debug?

orivej, 2017-12-01:
@fhunleth Thank you for your effort! You can run `nix-shell '<nixpkgs>' -A fwup` to enter the build environment with all the dependencies for `fwup`.

fhunleth, 2017-12-02:
I just ran the following in a fwup source directory:
```
nix-shell '<nixpkgs>' -A fwup
./autogen.sh
./configure
make
make check
```
All tests passed for me.

I'm running NixOS 17.09.2249.6b8833df272 (x86_64-linux) in VirtualBox. Are you running something different?

jtojnar, 2017-12-02:
Sorry about that, NixOS 17.09 contains the old version of `fwup`. Could you run `nix-channel --add https://nixos.org/channels/nixpkgs-unstable` and `nix-channel --update` before entering the shell.

fhunleth, 2017-12-02:
I checked out fwup v0.18.0 from Github to ~/fwup and built it there. I ran the commands that you wrote and re-entered the shell. Should I be compiling from a different directory?

jtojnar, 2017-12-02:
My bad, that should be fine. Then it looks like a problem with environment. Could you try using `nix-shell '<nixpkgs>' -A fwup --pure` to sandbox the shell?

fhunleth, 2017-12-02:
Ok, `--pure` is really cool.

I ran `make distclean` first to start a fresh build and then repeated what I did above. I have the same result. I.e. everything works.

orivej, 2017-12-02:
Sorry, we have not checked if the tests of the newer `fwup` fail in NixOS 17.09, only in the NixOS master (aka unstable). One of the differences is the upgrade to Glibc 2.26.

orivej, 2017-12-02:
Could you unpack https://github.com/NixOS/nixpkgs/archive/master.tar.gz somewhere and run `nix-shell path/to/master -A fwup --pure`? This should reproduce test failure.

fhunleth, 2017-12-02:
Downloaded and unpacked and I'm finally getting the failures! Investigating...

fhunleth, 2017-12-02:
Fixed! The difference was indeed related to glibc 2.26. The implementation of open() was changed to call the openat syscall. I didn't handle the openat syscall in the syscall validator, so it was throwing an error. It's handled now. Thanks for reporting this. I'll make a patch release shortly.

orivej, 2017-12-02:
Thank you for the fix (and for introducing me to the `open_by_handle_at` syscall)!

closed 2017-12-02 by fhunleth

fhunleth, 2017-12-02:
Thanks @jtojnar and @orivej again for reporting the issue and helping me reproduce it. The 0.18.1 release is now available with the fix.

commit 4164d02b (2017-12-02): verify_syscalls: handle openat()

This fixes an issue where the C library called the openat syscall. This
was ignored and verify_syscalls would incorrectly fail.

Fixes #83