fwup-home/fwup #85: Arch Linux AUR not working  because it isnt passing all testts

gallexme, 2018-02-02:


fhunleth, 2018-02-02:
I think that you're right that there are missing dependencies. Since the FAT filesystem regression tests are all failing, I'd say that `mtools` isn't installed. Here's a list: https://github.com/fhunleth/fwup/blob/master/docs/build_linux.md#common-dependencies. It bothers me a little that a couple of the sparse regression tests are failing. That might be something I need to actually fix. If those continue to fail, could you let me know what filesystem you're using or if you're on a RAID partition?

liberza, 2018-02-02:
I just corrected the dependencies in the AUR. Try installing again, it should get them automatically.

gallexme, 2018-02-02:
installed mtools manually and it worked(before u updated the aur)

tcoopman, 2018-03-09:
I'm trying to install this aur package and now other tests fail:

```
PASS: 001_simple_fw.test
PASS: 002_resource_subdir.test
PASS: 003_write_offset.test
PASS: 004_env_vars.test
PASS: 005_define.test
PASS: 006_metadata.test
PASS: 007_mbr.test
PASS: 008_partial_mbr.test
PASS: 009_metadata_cmdline.test
PASS: 010_fat_mkfs.test
PASS: 011_fat_setlabel.test
PASS: 012_fat_write.test
PASS: 013_fat_mv.test
PASS: 014_fat_cp.test
PASS: 015_fat_bad_sha2.test
PASS: 016_raw_bad_sha2.test
PASS: 017_fat_rm.test
PASS: 018_numeric_progress.test
PASS: 019_quiet_progress.test
PASS: 020_normal_progress.test
PASS: 021_create_keys.test
PASS: 022_signed_fw.test
PASS: 023_missing_sig.test
PASS: 024_metadata_sig.test
PASS: 025_bad_sig.test
PASS: 026_sign_again.test
PASS: 027_fat_timestamp.test
PASS: 028_osip.test
PASS: 029_5G_offset.test
PASS: 030_2T_offset.test
PASS: 031_fat_mkfs_5G.test
PASS: 032_fat_2T_offset.test
PASS: 033_bad_host_path.test
PASS: 034_missing_host_path.test
PASS: 035_streaming.test
PASS: 036_streaming_signed_fw.test
PASS: 037_streaming_bad_sig.test
PASS: 038_write_15M.test
PASS: 039_upgrade.test
PASS: 041_version.test
PASS: 042_fat_am335x.test
PASS: 043_fat_touch.test
PASS: 044_require_file_exist.test
PASS: 045_legacy_require_partition1.test
PASS: 046_unknown_create_fails.test
PASS: 047_unknown_apply_succeeds.test
PASS: 048_assert_size_less_than_success.test
PASS: 049_assert_size_less_than_fail.test
PASS: 050_assert_size_greater_than_success.test
PASS: 051_assert_size_greater_than_fail.test
PASS: 052_file_concatenation.test
PASS: 053_framed_progress.test
PASS: 054_framed_metadata.test
PASS: 055_metadata_partial_file.test
PASS: 056_list_tasks.test
PASS: 057_list_tasks_empty.test
PASS: 058_framed_list_tasks.test
PASS: 059_framed_error.test
PASS: 060_framed_streaming.test
PASS: 061_framed_partial_metadata.test
PASS: 062_long_meta_conf.test
PASS: 063_detect.test
PASS: 064_mbr_bootcode.test
PASS: 065_fat_mkdir.test
PASS: 066_fat_attrib.test
PASS: 067_usage.test
PASS: 068_readonly_output_error.test
PASS: 069_define_bang.test
PASS: 070_bad_mbr_offset.test
PASS: 071_big_mbr_offset.test
PASS: 072_compression_works.test
PASS: 073_multistep_fat.test
PASS: 074_fat_cache_fail.test
PASS: 075_big_fat_fs.test
PASS: 076_utf8_metadata.test
PASS: 077_fat_empty_file.test
PASS: 078_partial_mbr2.test
PASS: 079_uboot_setenv.test
PASS: 080_uboot_empty_env.test
PASS: 081_uboot_clearenv.test
PASS: 082_uboot_upgrade.test
PASS: 083_uboot_unsetenv.test
PASS: 084_corrupt_uboot.test
PASS: 085_raw_memset.test
PASS: 086_math.test
PASS: 087_crypto_compat.test
PASS: 088_missing_hash.test
PASS: 089_dev_null.test
PASS: 090_sparse_write.test
PASS: 091_sparse_write2.test
PASS: 092_sparse_write3.test
PASS: 093_sparse_concat.test
PASS: 094_sparse_concat2.test
PASS: 095_sparse_apply.test
PASS: 096_missing_resource.test
PASS: 097_reproduceable.test
PASS: 098_sparse_empty.test
PASS: 099_sparse_lasthole.test
PASS: 100_easy_option.test
PASS: 101_progress_range.test
PASS: 102_multi_requires.test
PASS: 103_error_msg.test
PASS: 104_info_msg.test
FAIL: 105_require_path_on_device.test
PASS: 106_large_length_field.test
PASS: 107_string_resource.test
PASS: 108_resource_order.test
PASS: 109_includes.test
PASS: 110_compression_level.test
PASS: 111_streaming_exit_fast.test
PASS: 112_fat_rm_missing_file.test
PASS: 113_fat_mv_error_cases.test
PASS: 114_fat_cp_error_cases.test
PASS: 115_fat_mkdir_error_cases.test
PASS: 116_fat_setlabel_error_cases.test
PASS: 117_on_error.test
PASS: 118_fat_touch_error_cases.test
PASS: 119_require_fat_file_match.test
PASS: 120_corrupt_fat.test
PASS: 121_missing_file.test
PASS: 122_missing_file_metadata.test
PASS: 123_double_fat_write.test
PASS: 124_uboot_bad_param.test
PASS: 125_trimmed_upgrade.test
PASS: 126_uboot_recover.test
PASS: 127_extra_metadata.test
PASS: 128_blank_offset.test
PASS: 129_mbr_overflow.test
PASS: 130_media_sizes.test
PASS: 131_raw_key.test
PASS: 132_key_as_param.test
PASS: 133_path_write_file.test
PASS: 134_pipe_write_file.test
PASS: 135_execute.test
PASS: 136_fat_overwrite.test
PASS: 137_path_write_sparse.test
PASS: 138_pipe_write_sparse.test
PASS: 139_mbr_signature.test
PASS: 140_trim.test
PASS: 141_exit_handshake.test
PASS: 142_file_resource_size.test
PASS: 143_include_path.test
PASS: 144_require_version.test
FAIL: 145_require_path_at_offset.test
============================================================================
Testsuite summary for fwup 1.0.0
============================================================================
# TOTAL: 144
# PASS:  142
# SKIP:  0
# XFAIL: 0
# FAIL:  2
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to https://github.com/fhunleth/fwup/issues
============================================================================
```

Any idea if I'm missing something as well?

tcoopman, 2018-03-09:
Found my problem, make check needs to be run as root.

fhunleth, 2018-03-09:
Ugh. I really didn't want the tests to require root. I guess those tests need to be smarter to avoid that permission denied error.

fhunleth, 2018-03-09:
Just to add, those tests that failed won't impact you at all when using `fwup`. They're trying to verify some things used when doing software updates on-device. However, if you have a minute or two to experiment with passing options to `find` to skip those files and avoid the error, that would be really helpful. Or if you could just do an `ls -las /dev/vboxusb`. I'm curious what the permissions are.

tcoopman, 2018-03-09:
Here you go

```
> sudo ls -las /dev/vboxusb
total 0
0 drwxr-x---  4 root vboxusers   80 Mar  9 19:25 .
0 drwxr-xr-x 22 root root      3500 Mar  9 20:16 ..
0 drwxr-x---  2 root vboxusers  120 Mar  9 19:24 001
0 drwxr-x---  2 root vboxusers   40 Mar  9 20:10 003
```

fhunleth, 2018-03-11:
I had the same `/dev/vboxusb` directory and was able to reproduce the issue by removing myself from the `vboxusers` group. I just needed to ignore the error from `find`. Nothing in the `vboxusb` was of interest to the regression test script anyway, so this is safe.

closed 2018-03-11 by fhunleth