fwup-home/fwup #113: metadata tests failing

liberza, 2019-11-16:
Hi! I was trying to update the Arch packages [fwup](https://aur.archlinux.org/packages/fwup/) and [fwup-git](https://aur.archlinux.org/packages/fwup-git/) today, but I'm getting similar failing tests for both. I haven't gotten around to updating these packages in a while, so perhaps they're missing a new dependency? test-suite.log files are below for both. Please let me know if you need any other info.

[test-suite-1.4.0.log](https://github.com/fhunleth/fwup/files/3854600/test-suite-1.4.0.log)
[test-suite-master.log](https://github.com/fhunleth/fwup/files/3854601/test-suite-master.log)

fhunleth, 2019-11-16:
Thanks for the heads up. Let me check...

fhunleth, 2019-11-16:
Cool, Arch is doing reproducible builds. Let me fix those tests.

fhunleth, 2019-11-16:
Thanks! I was able to reproduce and I fixed it in #114.

Assuming CI is happy and nothing else fails, I plan on making a v1.5.0 release tonight or tomorrow.

liberza, 2019-11-16:
Nice, thank you for the quick fix!

commit e7dff013 (2019-11-16): Use SOURCE_DATE_EPOCH in metadata check tests

This fixes an issue that was introduce when fwup started preferring
SOURCE_DATE_EPOCH over NOW for making deterministic archives. Metadata
tests had used NOW, but that was ignored on build systems that were
actually using SOURCE_DATE_EPOCH.

The fix is to temporarily override SOURCE_DATE_EPOCH so that the unit
tests work. This verifies that SOURCE_DATE_EPOCH works. Additionally,
009_metadata_cmdline.test has an additional test now that verifies that
the old NOW variable continues to work.

Fixes #113.
PR #114 (2019-11-16): Use SOURCE_DATE_EPOCH in metadata check tests
This fixes an issue that was introduce when fwup started preferring
SOURCE_DATE_EPOCH over NOW for making deterministic archives. Metadata
tests had used NOW, but that was ignored on build systems that were
actually using SOURCE_DATE_EPOCH.

The fix is to temporarily override SOURCE_DATE_EPOCH so that the unit
tests work. This verifies that SOURCE_DATE_EPOCH works. Additionally,
009_metadata_cmdline.test has an additional test now that verifies that
the old NOW variable continues to work.

Fixes #113.
closed 2019-11-16 by fhunleth