fwup-home/fwup #118: gpt checksum problem

deblauwetom, 2020-01-16:
Hello,

I can create gpt partitions using fwup, but gdisk and gparted always tell me there is a problem with the checksum:

```
GPT fdisk (gdisk) version 1.0.3

Caution! After loading partitions, the CRC doesn't check out!
Warning! Main partition table CRC mismatch! Loaded backup partition table
instead of main partition table!

Warning! One or more CRCs don't match. You should repair the disk!

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: damaged

****************************************************************************
Caution: Found protective or hybrid MBR and corrupt GPT. Using GPT, but disk
verification and recovery are STRONGLY recommended.
****************************************************************************

```
But the list of partitions is ok actually. Is this a known issue?

fhunleth, 2020-01-16:
Thanks for reporting this. It is not known and it's surprising since the unit tests compare checksums against sfdisk's calculations. 

Anything special to reproduce?

deblauwetom, 2020-01-16:
I am trying to make a "pingpong" firmware for the nvidia jetson nano. I already made this for the raspberry pi3 based upon the nerves project example. So I start with an uboot-environment, maybe that is different? Also, there are 12 small partitions I need to make, maybe that is also a bit special?

```
uboot-environment uboot-env {
    block-offset = ${UBOOT_ENV_OFFSET}
    block-count = ${UBOOT_ENV_COUNT}
}

# This firmware task writes everything to the destination media
task complete {
    # Only match if not mounted
    require-unmounted-destination = true

    on-init {

        gpt_write(gpt-a)
        
        uboot_clearenv(uboot-env)
        uboot_setenv(uboot-env, "fw_active", "a")
    }
    #here are my on-resource statements 
}

fhunleth, 2020-01-16:
Oh, I think the U-Boot environment block is overwriting the end of the GPT header. The GPT header goes from block 0 to 34. 

Could you try moving UBOOT_ENV_OFFSET to 34 or later?

12 partitions should be fine.

deblauwetom, 2020-01-16:
🎉 that was it!  I now moved it to 64 to be sure.
Also, this was the reason it only booted onto the recovery bootloader, probably it did not find the root filesystem due to the corrupt gpt. Thanks!

closed 2020-01-16 by deblauwetom

fhunleth, 2020-01-16:
Great!

Btw, I know a few people who would be very interested in a Jetson Nano port, so please let me know when you get something working or have something to share!

deblauwetom, 2020-01-16:
Hello, well, now it works to flash a sdcard with fwup to boot on the jetson nano. Unfortunately I can't give you my exact source code, I am doing this for my employer... However, it is really just building a distro for the nano with yocto (meta-tegra layer) and then checking the "....sdcard" file with "gdisk" to see the partition layout. Then just create the 12 partitions with their names, and just rawwrite the files you get in yocto to these partitions in fwup.

fhunleth, 2020-01-16:
Makes sense. Thanks for the info!