fwup-home/fwup #120: Handle new line characters in `--public-key`

coop, 2020-02-20:
New line characters are not currently supported when passed to `--public-key`.

Initially I had trouble diagnosing the issue because [fwup errors were not being correctly reported in elixir](https://github.com/ConnorRigby/elixir-fwup/pull/9). Once I understood the problem @jjcarstens helped me track down the issue to new line characters in our public keys:

```
iex(app@nerves-aers4klk7fbzr3q.local)3> NervesHubLink.Certificate.fwup_public_keys
["wFiHt2iyFDo980F9MuDzRRR3P7I4VHrKXiwQbqDq/2Y=",
 "0LpD4UWYDzjHBUCTfG7ADFAgUx2LKe4gqkQG8tiX7TY=\n",
 "kJ4SDTkJZ6OCBrFwNzZbqF3KcK02pBeH3eRZJFy/e18="]
```

Even though the firmware I was trying to deploy wasn't signed by the key with the trailing new line fwup still threw an error `"Error decoding public key"`. I believe that's because `NervesHubLink` [passes all the keys as arguments to fwup](https://github.com/nerves-hub/nerves_hub_link/blob/master/lib/nerves_hub_link/http_fwup_stream.ex#L67-L70).

Thanks for fwup!

fhunleth, 2020-02-20:
Good catch! This seems like an easy fix. I make it on my next pass through fixing fwup issues.

fhunleth, 2020-04-29:
Closed since this appears to have been fixed when the base64 code was changed. Commit b4d5593 adds a test in case it reappears.

closed 2020-04-29 by fhunleth