fwup-home/fwup #141: Fail if meta-product is not correct

davidepianca98, 2020-05-06:
Hello,
I am using fwup in framing mode to update the firmware on a device, sending every chunk I receive from a TCP connection through a pipe to the fwup process. It works perfectly but I would like to include a check by meta-product, so that it's not possible to flash a firmware for another device. If I had the whole file on the filesystem I could then run fwup -m to get the meta and then check if it matches, but as I'm applying the update on the fly I don't think it's possible.
Is it already possible with the current version?
Would it otherwise be possible to implement an additional option to pass on the command line with the value to check for, so that if it doesn't match, it would fail?

fhunleth, 2020-05-06:
Yes, a check could be added. I'll take a look next time I make a pass on `fwup` maintenance - probably in the next couple weeks.

Here are some other possibilities for now:

1. Firmware metadata is stored at the very beginning of the `.fw` file, so you only need a little data run `fwup -m`. This unit test is an example: https://github.com/fhunleth/fwup/blob/master/tests/055_metadata_partial_file.test. You could still stream the whole file through `fwup`, but on the side, you'd have code that runs after 512 bytes (or 1K since it's not like this has to be as small as possible) is received, saves those first bytes to a file and then runs `fwup -m` on that file.

2. If the currently running firmware's metadata can be stored in a uboot-formated environment block. You don't need U-Boot. This is just the format, you can check it with `require-uboot-variable`. Here's an example: https://github.com/nerves-project/nerves_system_rpi0/blob/master/fwup.conf#L329-L330. To give a nice error message when there's a mismatch, this code was added: https://github.com/nerves-project/nerves_system_rpi0/blob/master/fwup.conf#L391-L403.

davidepianca98, 2020-05-07:
Thank you very much, I think solution number 1 would be good enough for me, but the optimal solution would be to use uboot require as the board runs from U-Boot. I cannot currently use it as I have redundant env on the board, so I'm writing a patch to add support for it in fwup. When it's ready and tested I will send a pull request here. And I think you can close this issue unless you think the option I proposed should be added anyway.

fhunleth, 2020-05-07:
Thanks - also, thanks for looking into redundant u-boot environment support. I had been putting that off. Even if you only have the updates for https://github.com/fhunleth/fwup/blob/master/src/uboot_env.c, I'd take it to slowly push things forward.

closed 2020-05-07 by fhunleth