fwup-home/fwup #154: v1.8.2 Autodetect of `dev/mmcblk0` fails due to `/sys/block/mmcblk0/removable` returning 0

tapickell, 2020-11-01:
On PopOs 20.10 (Ubuntu based for System76)
```
OS: Pop!_OS 20.10 x86_64 
Host: Oryx Pro oryp6 
Kernel: 5.8.0-7625-generic 
```

when using the builtin micro sd card reader
```
[53096.994161] mmc0: new ultra high speed SDR104 SDHC card at address 5048
[53096.994816] mmcblk0: mmc0:5048 Y004R 3.74 GiB 
[53097.000141]  mmcblk0: p1
```

the system sets the sd card to `/dev/mmcblk0`
```
❯ list /dev | grep mmcblk
brw-rw----  1 root disk    179,     0 Nov  1 13:19 mmcblk0
brw-rw----  1 root disk    179,     1 Nov  1 13:20 mmcblk0p1
```
the removable flag is set to 0
```
❯ cat /sys/block/mmcblk0/removable
0
```

The device is not detected by Fwup
```
❯ fwup -D

```

In the `lib/mmc_linux.c` file there is a check for this value to be non zero
https://github.com/fhunleth/fwup/blob/main/src/mmc_linux.c#L104

I am also attempting to file a bug with PopOs and or Ubuntu as I believe that the removable flag should return 1 instead.

Thank you for your help with this @fhunleth

tapickell, 2020-11-01:
It looks like in PopOs they are suing their own systemd
and according to this function they are expecting removable to > 0 for sd cards as well.
I have not found were the mounting of the sd card happens where setting of removable should happen yet.
https://github.com/pop-os/systemd/blob/ddb09573ebb7264538af3f77f16b5d1c252a8551/src/mount/mount-tool.c#L1210

fhunleth, 2020-11-01:
Thanks for the information. I feel like I should be able to reproduce this on my next pass at updating `fwup`. In the mean time, hopefully manually passing `-d mmcblk0` to force the MicroSD card to be used works for you.

greghgradwell, 2020-11-01:
I don't have much to add, but I'm on Pop!OS 18.04 with Fwup v1.8.1 and having the same issue. I've tried with a fresh SD card as well as with one that already had been used to burn a Nerves firmware image.  
 
```
OS: Pop!_OS 18.04 LTS x86_64  
Host: Meerkat meer4   
Kernel: 5.4.0-52-generic 
```

```  
➜ ls -al /dev | grep mmcblk  
brw-rw----   1 root   disk    179,     0 Nov  1 11:52 mmcblk0  
brw-rw----   1 root   disk    179,     1 Nov  1 11:52 mmcblk0p1  
brw-rw----   1 root   disk    179,     2 Nov  1 11:52 mmcblk0p2  
brw-rw----   1 root   disk    179,     3 Nov  1 11:52 mmcblk0p3  
```

```
➜ cat /sys/block/mmcblk0/removable  
0
```

fhunleth, 2021-01-22:
All - sorry for the long wait. I looked into this with the help of @mattludwigs who has a Pop!OS system. The issue is due to a change in the Linux kernel and you all hit it first. I have a fix in #162.

PR #162 (2021-02-07): Fix autodetection of built-in SDCard readers
The built-in SDCard device driver in Linux marks these drives a
non-removal. See the comment at
https://elixir.bootlin.com/linux/v5.8.18/source/drivers/mmc/core/block.c#L2335.

This change modifies the drive detection logic to consider all built-in
SDCard readers to hold removable block devices from `fwup`'s point of
view. Even if the Linux kernel maintainers view these devices as
non-removable, this change should be safe to nearly all `fwup` users
since it's rare to use an SDCard for a system drive. There's another
check, though, so if you use an S [... cut]
closed 2021-02-07 by fhunleth