fwup-home/fwup #155: Test Suite fails for v1.8.2 compile on NixOS

madisoncommunitycare, 2020-11-02:
When using a version override to install the latest version of fwup inside a nix-shell, compilation fails. 

```
======================================
   fwup 1.8.2: tests/test-suite.log
======================================

# TOTAL: 188
# PASS:  181
# SKIP:  3
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

SKIP: 105_require_path_on_device
================================

./105_require_path_on_device.test: /build/source/tests/common.sh: line 55: which: not found
The root filesystem doesn't map to one device file, so can't test for rdev
Test skipped
SKIP 105_require_path_on_device.test (exit status: 77)

SKIP: 111_streaming_exit_fast
=============================

./111_streaming_exit_fast.test: /build/source/tests/common.sh: line 55: which: not found
GNU coreutils' timeout command unavailable so skipping test.
Test skipped
SKIP 111_streaming_exit_fast.test (exit status: 77)

SKIP: 145_require_path_at_offset
================================

./145_require_path_at_offset.test: /build/source/tests/common.sh: line 55: which: not found
The root filesystem doesn't map to one device file, so can't test for rdev
Test skipped
SKIP 145_require_path_at_offset.test (exit status: 77)

FAIL: 172_delta_upgrade
=======================

./172_delta_upgrade.test: /build/source/tests/common.sh: line 55: which: not found
100% [====================================] 2.85 KB in / 1.02 KB out
Success!
Elapsed time: 0.001 s
-- verify-syscalls report --
Calls to write(): 1
 Bytes written: 131072
 Averate bytes written/call: 131072.0 bytes/call
 Last offset written: 524288
Calls to read(): 1
 Bytes read: 0
 Averate bytes read/call: 0.0 bytes/call
----------------------------
./172_delta_upgrade.test: line 49: xdelta3: not found

Test failed!

Leaving test work files in '/build/source/tests/work-172_delta_upgrade.test'
FAIL 172_delta_upgrade.test (exit status: 127)

FAIL: 173_bad_delta_upgrades
============================

./173_bad_delta_upgrades.test: /build/source/tests/common.sh: line 55: which: not found
100% [====================================] 2.85 KB in / 1.02 KB out
Success!
Elapsed time: 0.000 s
-- verify-syscalls report --
Calls to write(): 1
 Bytes written: 131072
 Averate bytes written/call: 131072.0 bytes/call
 Last offset written: 524288
Calls to read(): 1
 Bytes read: 0
 Averate bytes read/call: 0.0 bytes/call
----------------------------
1+0 records in
1+0 records out
512 bytes copied, 2.2305e-05 s, 23.0 MB/s
updating: data/rootfs.next
        zip warning: Local Entry CRC does not match CD: data/rootfs.next
 (stored 0%)
fwup: xdelta3 error: not a VCDIFF input
  0% [                                    ]
fwup: xdelta3 error: not a VCDIFF input
./173_bad_delta_upgrades.test: line 60: xdelta3: not found

Test failed!

Leaving test work files in '/build/source/tests/work-173_bad_delta_upgrades.test'
FAIL 173_bad_delta_upgrades.test (exit status: 127)

FAIL: 174_streamed_delta_upgrade
================================

./174_streamed_delta_upgrade.test: /build/source/tests/common.sh: line 55: which: not found
100% [====================================] 2.85 KB in / 1.02 KB out
Success!
Elapsed time: 0.000 s
-- verify-syscalls report --
Calls to write(): 1
 Bytes written: 131072
 Averate bytes written/call: 131072.0 bytes/call
 Last offset written: 524288
Calls to read(): 1
 Bytes read: 0
 Averate bytes read/call: 0.0 bytes/call
----------------------------
./174_streamed_delta_upgrade.test: line 51: xdelta3: not found

Test failed!

Leaving test work files in '/build/source/tests/work-174_streamed_delta_upgrade.test'
FAIL 174_streamed_delta_upgrade.test (exit status: 127)

FAIL: 175_sign_delta_upgrade
============================

./175_sign_delta_upgrade.test: /build/source/tests/common.sh: line 55: which: not found
Firmware signing keys created and saved to 'fwup-key.pub' and 'fwup-key.priv'

Distribute 'fwup-key.pub' with your system so that firmware updates can be
authenticated. Keep 'fwup-key.priv' in a safe location.
./175_sign_delta_upgrade.test: line 42: xdelta3: not found

Test failed!

Leaving test work files in '/build/source/tests/work-175_sign_delta_upgrade.test'
FAIL 175_sign_delta_upgrade.test (exit status: 127)

============================================================================
Testsuite summary for fwup 1.8.2
============================================================================
# TOTAL: 188
# PASS:  181
# SKIP:  3
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to https://github.com/fhunleth/fwup/issues
============================================================================
make[4]: *** [Makefile:861: test-suite.log] Error 1
make[4]: Leaving directory '/build/source/tests'
make[3]: *** [Makefile:969: check-TESTS] Error 2
make[3]: Leaving directory '/build/source/tests'
make[2]: *** [Makefile:1060: check-am] Error 2
make[2]: Leaving directory '/build/source/tests'
make[1]: *** [Makefile:753: check-recursive] Error 1
make[1]: Leaving directory '/build/source/tests'
make: *** [Makefile:488: check-recursive] Error 1
builder for '/nix/store/vh1cjh0j0il0k5wjhn30jdj7f3757abn-fwup-1.8.2.drv' failed with exit code 2
error: build of '/nix/store/vh1cjh0j0il0k5wjhn30jdj7f3757abn-fwup-1.8.2.drv' failed
 ```

The nix-shell I'm using looks like this: 

```
{ pkgs ? import (fetchTarball https://github.com/nixos/nixpkgs/archive/nixpkgs-unstable.tar.gz) {} }:

with pkgs;

let
  fwup = pkgs.fwup.overrideAttrs (drv: rec {
    version = "1.8.2";
    src = fetchFromGitHub {
      owner = "fhunleth";
      repo = "fwup";
      rev = "v1.8.2";
      sha256 = "1p0vpby3iscib20416m0m3va0zx9fzw3bm4fhsmf6yfg3514iym7";
    };
  });
in

mkShell {
  nativeBuildInputs = [ pkg-config ];
  buildInputs = [
    automake
    autoconf
    pkgs.beam.packages.erlangR23.elixir
    erlangR23
    file
    fwup
    git
    lxqt.lxqt-openssh-askpass
    inotify-tools
    openssh
    postgresql
    nodejs
    rebar3
    squashfsTools
    xdelta
    yarn
  ];
}
```

fhunleth, 2020-11-02:
This seems to be the error that's causing the failures:

```
xdelta3: not found
```

I've never used NixOS before. I see that you have `xdelta` listed above, but I don't know if that supplies the `xdelta3` binary.

jasonmj, 2020-11-02:
Following up here (with my correct Github account) based on our conversation in Slack. 

**I wrote:** 
"I also tried xdeltaUnstable because it appeared to include xdelta3: https://github.com/NixOS/nixpkgs/blob/nixos-unstable/pkgs/tools/compression/xdelta/unstable.nix#L55"

**Frank wrote:** 
"My guess would be to remove libsodium (not your problem, but also not required) and to add xdelta."