fwup-home/fwup #167: fsck.fat v4.2 failure

ejc123, 2021-03-03:
Firstly, I do realize fsck.fat is part of dosfstools, not fwup so this may not be the right place.  
In dosfstools v4.2, the read_boot() function errors when it can't access the last sector of the filesystem.  Previously, it checked, but didn't error.
All of the tests that use fsck.fat fail except **044_require_file_exist** which sets up a partition after the fat partition and writes to it.  I modified 010_fat_mkfs to do something similar and it started passing.
These are the failing tests:
`010_fat_mkfs.test
011_fat_setlabel.test
012_fat_write.test
013_fat_mv.test
014_fat_cp.test
017_fat_rm.test
031_fat_mkfs_5G.test
032_fat_2T_offset.test
042_fat_am335x.test
043_fat_touch.test
065_fat_mkdir.test
066_fat_attrib.test
073_multistep_fat.test
074_fat_cache_fail.test
075_big_fat_fs.test
077_fat_empty_file.test
117_on_error.test
136_fat_overwrite.test
146_deploy_variables.test
156_fat_truncation_bug.test`
The test-suite.log is attached
[test-suite.log](https://github.com/fwup-home/fwup/files/6078108/test-suite.log)

fhunleth, 2021-03-03:
Thanks for the report! It sounds like I got away with unit test sloppiness and dosfstools 4.2 caught me. It makes sense that what you did made the error go away. 

Let me take a quick look later. I think there should be a nice solution to this, but I don't have it now. 

What OS are you on that has dosfstools 4.2?

ejc123, 2021-03-03:
Arch Linux

fhunleth, 2021-03-03:
I should have bet money on that. You Arch people are always the first people to hit everything.

fhunleth, 2021-03-04:
Fixed in #168, #171, and #172. 

That new version of dosfstools turned out to have some pretty good checks. I think that it helps my confidence in the FAT support in fwup, so thanks for bringing it to my attention.

closed 2021-03-04 by fhunleth