fwup-home/fwup #174: question: sometimes, for ext4 partition, no sparse compression is done

deblauwetom, 2021-03-26:
Hello,

I am using fwup succesfully for a while, it's great! But now I noticed something with the size of the rootfs inside the fwup file. I have a local build, using yocto, and one on my buildserver(linux mint 19.3, now upgraded to linux mint 20). The ext4 rootfs is actually 1,6GB. And on my local build, the file inside the fwup archive is 1,2 GB, because sparse handling I think, so that is OK. But I build the same software on my buildserver, and there fwup archive contains the whole 1,6GB file. So I have no explanation for it. I tried the sparse check option, and it says sparse is enabled. I'm using 1.5.1 still, but tried with 1.8.4 and is still the same issue. So what can I check further? What could be a reason?

The resulting fwup files are actually both about the same size, but the problem is that I'm using the pingpong system and with the version from the buildserver I get problems upgrading, while with the version on my laptop it works perfectly. The reserved partition size is large enough to contain the full 1,6GB raw file, so I'm a bit puzzled.

I know it's a fuzzy issue, but maybe a hint could help me already :)
Best regards,
Tom,

fhunleth, 2021-03-27:
Hi Tom,

The sparse file feature is depends on what filesystem that you're using on the build server. The way it works is that `fwup` asks the OS where the holes in a file are and then saves that info to the archive. 

About a year ago, I started seeing some CI servers that have filesystems which were pretty unreliable with reporting the holes in sparse files. I never investigated since all of my work since then has been with compressed filesystems that didn't need the feature. I "think" that the common thread was that if the CI server was a virtual machine or a dedicated machine, it worked and if it was in a Docker container, it didn't. I'm not sure, though, and that doesn't make sense to me. My first guess would be to see what filesystem is in use on your build server and see if that provides a hint.

deblauwetom, 2021-04-07:
Hello,

Thanks for the reply. It seems somehow, the file is not sparse, and it's on a pure ext4 filesystem. I am using yocto, and it seems on my buildserver, I have the file with a timestamp, and one without the timestamp. I discovered a tool called fallocate, which is on linux, and you can recreate a file to be sparse. So I tried with "fallocate -d rootfs.img -v" on the rootfs.img with a timestamp, and it only detected 32mb of sparse space. And I tried with the non-timestamped version, and that could detect 400 mb of sparse space. When I now run fwup using the non-timestamped version, then fwup detects correctly that it is a sparse file and the fwup data directory contains the 1,2 GB file instead of a 1,6 GB file.
So that hopefully fixes it for me now 🤞

deblauwetom, 2021-04-08:
Hi, it seems that fixed it for me.

closed 2021-04-08 by deblauwetom

fhunleth, 2021-04-08:
Interesting. Thanks for following up.