fwup-home/fwup #181: fwup:amd64 : Depends: libc6:amd64 (>= 2.4) but it is not installable

ayesustenir, 2021-08-29:
### Issue
I try to install `fwup` for nerves project with docker like that:
```
FROM elixir:1.12

RUN apt-get update && \
    apt-get install -y build-essential automake autoconf git squashfs-tools ssh-askpass pkg-config curl && \
    rm -rf /var/lib/apt/lists/*

ENV FWUP_VERSION="1.8.4"
RUN wget https://github.com/fwup-home/fwup/releases/download/v${FWUP_VERSION}/fwup_${FWUP_VERSION}_amd64.deb && \
    apt-get install -y ./fwup_${FWUP_VERSION}_amd64.deb && \
    rm ./fwup_${FWUP_VERSION}_amd64.deb && \
    rm -rf /var/lib/apt/lists/*

RUN mix local.hex --force && \
    mix local.rebar --force && \
    mix archive.install hex nerves_bootstrap 1.10.2 --force  && \
    mix archive.install hex phx_new 1.5.9 --force
```

But, I got the following error.
```
#7 1.375 Resolving github-releases.githubusercontent.com (github-releases.githubusercontent.com)... 185.199.111.154, 185.199.110.154, 185.199.108.154, ...
#7 1.467 Connecting to github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.111.154|:443... connected.
#7 1.639 HTTP request sent, awaiting response... 200 OK
#7 2.083 Length: 324610 (317K) [application/octet-stream]
#7 2.083 Saving to: ‘fwup_1.8.4_amd64.deb’
#7 2.087
#7 2.087      0K .......... .......... .......... .......... .......... 15%  604K 0s
#7 2.168     50K .......... .......... .......... .......... .......... 31%  650K 0s
#7 2.243    100K .......... .......... .......... .......... .......... 47% 1.29M 0s
#7 2.281    150K .......... .......... .......... .......... .......... 63% 1018K 0s
#7 2.330    200K .......... .......... .......... .......... .......... 78% 2.04M 0s
#7 2.354    250K .......... .......... .......... .......... .......... 94% 2.36M 0s
#7 2.375    300K .......... .......                                    100%  890K=0.3s
#7 2.394
#7 2.396 2021-08-29 05:08:31 (1022 KB/s) - ‘fwup_1.8.4_amd64.deb’ saved [324610/324610]
#7 2.396
#7 2.416 Reading package lists...
#7 2.443 Building dependency tree...
#7 2.444 Reading state information...
#7 2.447 Some packages could not be installed. This may mean that you have
#7 2.447 requested an impossible situation or if you are using the unstable
#7 2.447 distribution that some required packages have not yet been created
#7 2.447 or been moved out of Incoming.
#7 2.447 The following information may help to resolve the situation:
#7 2.447
#7 2.447 The following packages have unmet dependencies:
#7 2.448  fwup:amd64 : Depends: libc6:amd64 (>= 2.4) but it is not installable
#7 2.448 E: Unable to correct problems, you have held broken packages.
------
executor failed running [/bin/sh -c wget https://github.com/fwup-home/fwup/releases/download/v${FWUP_VERSION}/fwup_${FWUP_VERSION}_amd64.deb &&     apt-get install -y ./fwup_${FWUP_VERSION}_amd64.deb &&     rm ./fwup_${FWUP_VERSION}_amd64.deb &&     rm -rf /var/lib/apt/lists/*]: exit code: 100
```
Could anyone point out how to solve the issue?
### Environment
- Docker version 20.10.8, build 3967b7d
- Mac OS Big Sur (M1)

fhunleth, 2021-08-29:
Wouldn't Docker on an M1 require an `aarch64` package rather than an `amd64` one? Or does Docker run on QEMU to emulate x86 on the M1?

If an `aarch64` package is needed, then you'll need to build from source for now. I don't have `aarch64` packages yet.

ayesustenir, 2021-08-30:
@fhunleth , yes I think we need `aarch64`. I just tried to run `elixir` image without fwup installation process. I tried to print out `uname -a` and I got:
```
Linux c1643bf93cf2 5.10.47-linuxkit #1 SMP PREEMPT Sat Jul 3 21:50:16 UTC 2021 aarch64 GNU/Linux
```

May be I will try to build from source . Could you have some reference link how to build from source, please let me know?

fhunleth, 2021-08-30:
To build from source, download the [source tarball](https://github.com/fwup-home/fwup/releases/download/v1.9.0/fwup-1.9.0.tar.gz) and look at [Linux build instructions](https://github.com/fwup-home/fwup/blob/main/docs/build_linux.md#tldr).

ayesustenir, 2021-08-31:
Thanks @fhunleth, I followed the instructions for [Linux build instructions](https://github.com/fwup-home/fwup/blob/main/docs/build_linux.md).Finally, I successfully installed it. 

[image]

My docker build instructions is :
```
FROM elixir:1.12

RUN apt-get update && \
    apt-get install -y build-essential automake autoconf git squashfs-tools ssh-askpass pkg-config curl && \
    apt-get install -y libtool mtools unzip zip help2man libconfuse-dev libarchive-dev xdelta3 && \
    rm -rf /var/lib/apt/lists/*

ENV FWUP_VERSION="1.8.4"
RUN wget https://github.com/fwup-home/fwup/releases/download/v${FWUP_VERSION}/fwup-${FWUP_VERSION}.tar.gz && \
    apt-get install -y libtool mtools unzip zip help2man libconfuse-dev libarchive-dev xdelta3 && \
    tar -xvzf fwup-${FWUP_VERSION}.tar.gz && \
    cd fwup-${FWUP_VERSION} && \
    ./configure && \
    make -j4 && \
    # make check && \
    make install && \
    rm /fwup-${FWUP_VERSION}.tar.gz
```
Do you notice I commented the `make check` step? I found this step is failing for now. I don't know what is the issue it is. It is failing 20 tests.

[image]

But, I just skipped `make check` step and it is successfully installed using docker on my M1 Mac.
Thanks you for your support @fhunleth.

fhunleth, 2021-08-31:
Thanks for the update. There are several unit tests that don't work under Docker or on the default MacOS filesystem. I'm guessing that those are the ones that failed and the test infrastructure didn't skip them. Could you post the `test-suite.log` or the list of failed tests just in case, though?

ayesustenir, 2021-08-31:
Sure thing @fhunleth , here is `test-suite.log` file result:

[test-suite.log](https://github.com/fwup-home/fwup/files/7083367/test-suite.log)

fhunleth, 2021-08-31:
Thanks! It turns out that the failures aren't anything fancy. It's just that `fsck.fat` isn't installed. That can be installed with `apt install dosfstools`. However, it's only needed to run the unit tests and not to use `fwup`, so there's no need for you to add it.

closed 2021-08-31 by fhunleth

ayesustenir, 2021-08-31:
thanks @fhunleth for your detail explanation.