fwup-home/fwup #191: Support usb drive or card reader

mohsen-in, 2022-07-03:
Hello

First, thank you for your efforts
In windows, I can't write on usb or card reader
It does not recognize anything at all

```
> fwup -z
fwup: No memory cards found. Try reinserting the card.
```

Thanks

fhunleth, 2022-07-03:
Thanks for letting me know about this. Can you tell me more about the version of Windows that you're using, whether you're using a USB or built-in card reader? Also, do you know the path to the MicroSD card that `fwup` should have detected?

mohsen-in, 2022-07-03:
System : Windows 11 Version 10.0.22000
Micro SD : PNY 32GB Class 10
Card reader : Oscar
USB 3.0

I don't know the sd card path for fwup, but I know my F:\ drive belongs to sd card

fhunleth, 2022-07-03:
Thanks. I'll try to reproduce. Since this might take a while, you should be able to get past this by specifying `-d microsd.img` where the `microsd.img` can be any file name. `fwup` will write the image to the file and then you can use Etcher or another program to write the img file to the MicroSD card.

mohsen-in, 2022-07-03:
excellent
Thank you