fwup-home/fwup #200: Handle /dev/null as a special case output device

RossiSoftLtd, 2023-02-27:
Sometimes you have 'fw' bundles that also contain one or more tasks which not necessarily require to write on an eMMC device but only exists to allow processing files via an external tool (i.e. --unsafe option together with path_write and execute commands).

The issue is that because whilst invoking the FWUP tool with the apply command you always need to specify a device name so for such cases the device doesn't really make sense, therefore the only option is to use a fake filename that will be created on the local file system with a zero size length and that is a little bit ugly.

I reckon allowing the FWUP tool to accept the '/dev/null', which is currently rejected, as a possible candidate to support such scenarios would be preferable and neat than a fake filename?

Thanks in advance and please keep up the good work, this tool is a fundamental component on our projects.

Kind regards
Eddy

fhunleth, 2023-02-28:
Thanks for all of the details on your use case. I understand what you're doing and it makes sense. I'm trying to set aside some time in the next few weeks to work on several fwup improvements. If you can wait that long, I think this will be an easy add for me.

RossiSoftLtd, 2023-02-28:
Thanks for the quick reply. Unfortunately we need this earlier so we are going to make a patch but looking forward to your solution which I'm sure will be proper and neater, also we can't wait for the new feature/improvement you just mentioned ;-)

Kind regards
Eddy

RossiSoftLtd, 2023-02-28:
[handle-dev-null-as-a-special-case-output-device.patch](https://github.com/fwup-home/fwup/files/10848936/handle-dev-null-as-a-special-case-output-device.patch)

This is our very simple patch that works in our case scenario; obviously it's not 100% ideal, but enough to keep us going whilst awaiting for your proper solution.

Kind regards
Eddy

fhunleth, 2023-02-28:
Awesome! Well, that makes it even easier for me. Thanks for sharing the patch.

closed 2023-03-22 by fhunleth