fwup-home/fwup #201: Support delta-source-fat-offset and delta-source-fat-path

jeanparpaillon, 2023-03-13:
Delta upgrades do not support applying delta on FAT files, zImage on boot partition for many devices, precisely.

Supporting delta-source-fat-offset and delta-source-fat-path is needed for applying delta upgrades on FAT files.

closed 2023-03-23 by fhunleth

jeanparpaillon, 2023-03-23:
@fhunleth thanks for looking at it. 
With my version of the code, I got the following error when trying to apply a patch:
"target window checksum mismatch"
The condition for reaching this message is quite obscure for me (https://github.com/fwup-home/fwup/blob/main/src/3rdparty/xdelta3/xdelta3-decode.h#L799 )
Any idea on what may be wrong ?

fhunleth, 2023-03-23:
Whenever I've gotten an xdelta3 error, it's either been that I had a different source file on the target than what I thought that I should have had or a bug with reading the xdelta3 file, or source data. 

Is there any way I can take a look?

Also, I have no reason to believe that `main` would behave different than your version of the code, but did you try that?

fhunleth, 2023-03-29:
Great! Thank you for the update.

commit 4f97cab6 (2023-03-23): Add support for delta-source-fat-* directives

Closes #201