fwup-home/fwup #214: u-boot redundant environment support issue

RossiSoftLtd, 2023-06-12:
Hi,
I have recently started to use the u-boot redundant environment support for our projects and noticed that after some writes the content of such area wasn't consistent with what expected, even further I discovered that after such condition any subsequent write didn't work properly.

After further investigation I noticed the issue was happening after 127 writes and looking at the code I figured out what the issue was and hopefully fixed it.

In attachment the patch that according to my tests seem working OK, but please feel free to completely discard/change to your styling if you wish, I just thought to share as it might be useful to others.

Thanks for making this tool that I use extensively and enjoy its power and simplicity ;-)

Kind regards
Eddy

[fixes-handling-redundant-uboot-environment.patch](https://github.com/fwup-home/fwup/files/11724017/fixes-handling-redundant-uboot-environment.patch)

RossiSoftLtd, 2023-06-13:
No worries, regarding the patch, we are integrating 'fwup' using Yocto (Gatesgarth) and the patch compiles fine, however I just noticed today that integrating the patch with an older Petalinux-Xilinx it will fail when using the two 'static const' definitions because their use in a switch statement in 'C' is not allowed, albeit until C99 I guess, so I simply changed to a couple of horrible '#defines', but as I said before feel free to change to whatever you think works best ;-).

Kind regards
Eddy

fhunleth, 2023-07-17:
@RossiSoftLtd Sorry for the delay - something came up in life, but I'm back! I looked at the issue and opted for a slightly different fix. Your patch worked, but caused a different unit test to fail. That one had to do with recovering after U-Boot environment corruption. It probably was nothing - like the unit test being too picky. However, once I saw what you fixed and had a good test to verify the issue, I took a crack at a minimal change. 

If you're around and can take a quick look, let me know if you have any concerns. If I don't hear from you, I'll just push this out since I think that I've tested the case sufficiently and anyone using redundant U-Boot environments definitely needs it.

RossiSoftLtd, 2023-07-18:
Hi,
hope you're well and your problems solved ;-)

Regarding your changes have you checked accessing the u-boot compatible area via standard user space utilities like fw_printenv/fw_setenv as well? I'm saying that because such utilities which we are using in combination with 'fwup' are particularly strict on the swap buffer edge cases and our patch was carefully catering for that in selecting the right buffer.

fhunleth, 2023-07-18:
Thanks. That's a good suggestion.

The tests update the U-Boot environment 255 times to roll the counter. I just added a call to `fw_printenv` to verify each iteration to make sure that the expected buffer was read back by the U-Boot tools. It was.

If there's another case that you've tested that's not exercised, please let me know. Other unit tests exercise corruption recover which is, I think, the main thing that's not tested by unit test 200.

commit e1e74efe (2023-07-19): Fix redundant U-Boot env selection on wrap

An incorrect U-Boot environment was being selected at the 128th
generation. Thanks to @RossiSoftLtd for identifying the problem and
figuring out how to fix it.

This also adds a unit test to verify that the right environment is
selected after running fwup enough times to wrap the 8-bit counter.

Fixes #214
closed 2023-07-19 by fhunleth