fwup-home/fwup #216: UMS is not detected by fwup on Windows 11

JackShaver, 2024-01-02:
I am on Windows 11 using the latest version of fwup, installed via the executable.
When using .\fwup.exe -z, I'm getting:
fwup: No memory cards found. Try reinserting the card.

Now when I use Get-PhysicalDisk, I can view my Linux UMS disk as physical disk number 1.
So I try to use .\fwup.exe -d 1 .\firmware.fw where firmware.fw is my firmware.
This sort of works, it appears to work as intended but nothing is actually written to the MMC.
Additionally, the write time is just far too short compared to another computer running fedora and fwup (2 seconds instead of 2 plus minutes).

Im wondering if this is a windows issue, a physical drive number issue, or something else.
The target is an eMMC exported to the host PC as a USB mass storage device. Windows recognizes it and gives it a disk letter.
Additionally, the device is using u-boot to create the UMS.
This has been done before on a different PC, on Windows 10, but I'm unsure how they did it.

Thank you for your time!

fhunleth, 2024-01-03:
Oh no. I haven't used Windows 11 with `fwup` and don't have a Windows 11 machine handy. 

As an temporary workaround, you can create a disk image `fwup -d imagefile.img` and then use Etcher or another tool to write to the eMMC.

I think that I'll need help to fix Windows 11 and that might take a while. I hope that the workaround is ok for now.

JackShaver, 2024-01-03:
Ok thank you for the speedy response, Ill try it out.

JackShaver, 2024-01-03:
Update: 
Using the command .\fwup -d firmware.img .\firmware.fw works just fine, I realized what what was going on, and etcher is able to write to the UMS as well, and everything works on the device.
Additionally, when trying to use the command: 
.\fwup.exe -d \\.\PhysicalDrive1 .\firmware.fw
fwup is able to recognize this as device namespace but throws some errors, starting with:
"fwup: Error determining the size of \\.\PhysicalDrive1"
Do you know what I could do to fix this such that I can use fwup by itself?

For now, I'm going to pursue the workaround of creating the .img with fwup and then uploading it with etcher.
Additionally, there is a misleading error message about formatting when trying to use:
.\fwup.exe -d \\.\PhysicalDisk1 .\firmware.fw
"fwup: Target device must be formatted like \\.\Physical**Disk**# where ..."
The formatting actually prefers "... \\.\Physical**Drive**# .."

Thanks again for the help!

JackShaver, 2024-01-05:
Update 2:

Good news! fwup works just fine on Windows 11 and is able to program my UMS eMMC as expected.
The issue was I wasn't executing fwup with elevated privileges! 
It's pretty obvious now, because balena cli requires them too, but Im just glad I got it working, and that there was never an issue with fwup.

So don't worry, fwup works fine on Windows 11.

closed 2024-01-05 by JackShaver

fhunleth, 2024-01-05:
Thanks for letting me know that it works!

I also fixed the error message that you pointed out was wrong in ac340b44c4b95f3785981b20f112cf7094ae12ca.