fwup-home/fwup #221: Test failure installing 1.10.1

sardaukar, 2024-04-17:
I'm getting a single test failure with 1.10.1 installation using the `fwup-git` AUR package:

```
FAIL: 175_sign_delta_upgrade
============================

Firmware signing keys created and saved to 'fwup-key.pub' and 'fwup-key.priv'

Distribute 'fwup-key.pub' with your system so that firmware updates can be
authenticated. Keep 'fwup-key.priv' in a safe location.
updating: data/rootfs.next
	zip warning: Local Entry CRC does not match CD: data/rootfs.next
 (stored 0%)
updating: data/rootfs.next
	zip warning: Local Entry CRC does not match CD: data/rootfs.next
 (stored 0%)
Valid archive with a good signature
Valid archive with a good signature
Valid archive without a signature
fwup: ZIP data length mismatch for rootfs.next

Test failed!

Leaving test work files in '/home/sardaukar/.cache/yay/fwup-git/src/fwup/tests/work-175_sign_delta_upgrade.test'
FAIL 175_sign_delta_upgrade.test (exit status: 1)

```

Any ideas about what I could do to fix it?

sardaukar, 2024-04-17:
Might be related to https://github.com/libarchive/libarchive/issues/1550

fhunleth, 2024-04-20:
Thanks for letting me know about this. I see that libarchive 3.7.3 also has a release note that looks related. The `fwup` tests are very picky. I'm pretty sure everything is ok, but I'm going to leave this open until I get a chance to understand it.

fhunleth, 2024-04-26:
I just released 1.10.2. All unit tests pass now. I create #227 to track the regression, but thankfully it's a really niche case that may affect no one. I still want it fixed, but there's no reason for it to cause you or most others concern.

closed 2024-04-26 by fhunleth