fwup-home/fwup #222: Suggestion: Ignore tiny discovered drives/devices

lawik, 2024-04-19:
I have an audio interface plugged into my main machine and it exposes a very small mass storage device with a manual and maybe a driver on it. It shows in fwup as 0.0GiB. It would be neat if this was just disregarded. Maybe anything under 0.1GiB or similar?

fhunleth, 2024-04-19:
How big is the audio interface device? Could you run `fwup -D` to get a more precise size. Setting a lower bound definitely seems like a good idea. I just need to do some checking if it's set to 100 MiB since that vaguely seems like something I've seen.

lawik, 2024-04-19:
```
/dev/sdb,196608

Scarlett Welcome Disk (0.10)
Y8PPD2D0B5D4DA
197 KB (196,608 bytes)
```

fhunleth, 2024-04-20:
@lawik Could you try out the PR? Build instructions are at https://github.com/fwup-home/fwup/blob/main/docs/build_osx.md#building-and-installing-with-homebrew.

lawik, 2024-04-20:
Will do. But brave to assume I do my AV work on a Mac :D I am way more foolish and will test it on Linux thank you very mich 💖

PR #223 (2024-04-22): Do not autodetect very small drives
The size is set to 1 MiB, so it's much smaller than any use of fwup that
I've seen. It will help reduce false positives from devices that have
docs on a small read-only drive.

Fixes #222
closed 2024-04-22 by fhunleth