fwup-home/fwup #230: Red Hat support

jsimmonds2, 2024-07-22:
Just curious why RHEL and Fedora are such difficult platforms to build **fwup** on ?

fhunleth, 2024-07-22:
I can't imagine it's hard. The tool that I had been using, `fpm` stopped working. I had to remove it and it looked like the best I could do was manually create the packages. See 103163237197b9a7159a59faef43f00e3f446b92.

It's not really practical for me to maintain packages for everyone's favorite distribution, so I do the best that I can. PRs are welcome and appreciated.

jsimmonds2, 2024-07-23:
Thanks @fhunleth . . I apologise for my petulant voice in the OP.
It would be better for me to do the work and offer an RPM installer.

Your time is certainly stretched. I'll build from sources on my Fedora 40 WS edition and be happy with that.

fhunleth, 2024-07-23:
No apologies needed, but I appreciate the sentiment. I know the frustration when features disappear due to bit rot.

I hope you don't mind me leaving this issue open until we get an RPM again. Maybe this will encourage someone to help. Or if no one is around, on my next maintenance pass on `fwup`, I'll see if ChatGPT can offer something that can get us there.

jsimmonds2, 2024-08-28:
Hello @fhunleth I recently built **fwup** on Fedora 40 (adm64) with success . . and it was painless.
```bash
sudo dnf install @development-tools squashfs-tools libarchive-devel libconfuse-devel
```
then use the documented make procedure for Linux systems.

Please close this issue at your disgression . . I think my OP was more a gripe about 'ready-made' fwup package than a DIY exe.
:heart:

fhunleth, 2024-08-28:
Thanks for the update. I'm glad it works. It should be painless, but I agree that ready-made packages are much easier. I'm going to leave this open. I still hope to find someone to help make a RedHat package or see something that makes it look easy for me as a non-RedHat user to make it.