fwup-home/fwup #251: 130_media_sizes.test fails on my machine during make check

ponkotuy, 2025-06-15:
## Description

When running make check, the test 130_media_sizes.test fails on my machine. I was able to proceed with installation by skipping make check, so this is not urgent, but I wanted to report the issue.

## Details

From the logs, it appears that the failure occurs at the maximum size value of 7812499999. I have attached the relevant logs for reference.

## Environment

- fwup version: latest from main branch
- OS: ArchLinux
- Architecture: x86_64
- Compiler: gcc (GCC) 15.1.1 20250425

## Additional Information
- The issue is reproducible consistently on my machine.
- Please see the attached log for more details.

[130_media_sizes.log](https://github.com/user-attachments/files/20745884/130_media_sizes.log)

fhunleth, 2025-06-15:
Thank you for the issue report and especially letting me know your setup. I'll take a look soon.

fhunleth, 2025-06-15:
I tried to reproduce on Arch/x86_64 and the test passed for me. That particular test needs to create a very large sparse file. I wonder if there's some interaction with the filesystem and the test doesn't give a good error message. What filesystem are you using and how much free space is there?

In any case, I'm pretty sure this is a unit test issue.

ponkotuy, 2025-06-16:
I did some additional tests:
- On the same f2fs partition, I increased the free space to 143GB, but the result was the same — the test still failed.
- I ran the test on an NFS-mounted NAS (HDD: WDC WD80EAAZ-00BXBB0 8TB, filesystem: EXT4) from the same PC, and all tests passed.