fwup-home/fwup #253: fwup gets stuck on particular upgrade task (nerves_system_grisp2)

lawik, 2025-07-18:
This is a script for reproducing, it grabs my library that can generate deltas, based on how NervesHub does it.

```bash
#!/bin/bash
set -e

echo "Setting up project..."
yes | mix nerves.new --target=grisp2 foo

cd foo
export MIX_TARGET=grisp2

echo "Building..."
mix deps.get
mix firmware

echo "Generating image..."

fwup -a -d device.img -i /home/lawik/nervescloud/foo/_build/grisp2_dev/nerves/images/foo.fw -t complete
rm -fr ./tmp

echo "Generating delta..."
elixir --eval 'Mix.install([{:fwup_delta, github: "underjord/fwup_delta"}]); FwupDelta.do_generate("/home/lawik/nervescloud/foo/_build/grisp2_dev/nerves/images/foo.fw", "/home/lawik/nervescloud/foo/_build/grisp2_dev/nerves/images/foo.fw", Path.expand("./delta.fw"), Path.expand("./tmp"))'

echo "Applying delta..."
fwup -a -d device.img -i delta.fw -t upgrade --verbose
```

For me it sticks at 16.38Kb:

```
fwup: require-uboot-variable(uboot-env, nerves_fw_active) -> not met
fwup: require-uboot-variable(uboot-env, nerves_fw_active) -> met
fwup: require-uboot-variable(uboot-env, nerves_fw_validated) -> met
fwup: require-uboot-variable(uboot-env, a.nerves_fw_platform) -> met
fwup: require-uboot-variable(uboot-env, a.nerves_fw_architecture) -> met
fwup: Upgrading partition B
  2% [                                    ] 16.38 KB in / 524.29 KB out^C   
```

I have this (but Elixir code) as part of my tests with grisp2 commented out because it is the only one that hangs.

fhunleth, 2025-07-18:
Thanks. I'm able to reproduce.

lawik, 2025-07-18:
Phew